kakopappa/sinric #91: the device 'online' field always true problem

mrmojtabaa, 2018-07-18:
when i get status of my device from google , google request's 'action.devices.QUERY' intent from iot.sinrice.com , the response is something like this 

`{
    "requestId": "ff36a3cc-ec34-11e6-b1a0-64510650abcf",
    "payload": {
        "devices": {
            "123456789...": {
                "online": true,
                "on": true,
                "brightness": 100
            }
        }
    }
}`

but always the 'online' field is true , even the device not connected to power.

kakopappa, 2018-07-21:
sorry about the late response. it's fixed now

mrmojtabaa, 2018-07-21:
thanks a lot, 
but it has a little problem yet !
the 'online' and 'on' fields , have same result !
for example if device is online but i turned off it , the values should be 
`'online' : true ,
'on' : false
`

but now the result is 
`'online' : false
'on' : false
`

kakopappa, 2018-07-21:
You are right. now online status = on/off status which is wrong.  I misunderstood earlier. sorry about that

Now there is a new problem. Problem is 

1. ESP devices connect to sinric.com using the API key.
2. The server knows to which user account this ESP belongs to (api key to account id mapping).
3. Since the client does not send the device id when connecting, the server cannot check whether the device is connected or not.

We have to send the device id when connecting. Have to look at WebSocket connection headers in ESP whether we can pass the device id and how to pass it.

mrmojtabaa, 2018-07-22:
oh yes! you are right 
i searched in [arduinoWebSockets](https://github.com/Links2004/arduinoWebSockets)  in the [WebSocketsClient.h](https://github.com/Links2004/arduinoWebSockets/blob/master/src/WebSocketsClient.h) there is a method 
`void setExtraHeaders(const char * extraHeaders = NULL);`
and there is a sample for it,
`webSocket.setExtraHeaders("foo: I am so funny\r\nbar: not");`

i think we can use it like this ,
`webSocket.setExtraHeaders("deviceId: 123456...");`

it is from device , but i don't now how to fetch this data in server side ... !

DonKracho, 2018-08-03:
just an idea:
you implemented the function void setPowerStateOnServer(String deviceId, String value) in your example.

When the ESP device boots up it has to tell the status of each device id handled by the ESP to keep the Sinric server in sync. Can't we use this functon as heartbeat the ESP has to send in a time intervall of about 15 minutes? And If the Sinric server does not get this 'heartbeat' it handles the device as offline.

Then you wil have to specify the heatbeat time in order to keep the Sinric server in an acceptable traffic load.

mrmojtabaa, 2018-11-03:
Whats up @kakopappa ? is there any way ?