kakopappa/sinric #92: Android app improvements

kakopappa, 2018-07-21:
- Add different rooms.

- Refresh button on the app screen

DonKracho, 2018-07-22:
Great project! Thank you I want to control 433Mhz RF Switches via a WeMos D1 device.

For programming individual  RF outlets and the Master functionality (switching all devices at once) I need to have separated On and Of buttons instead of a toggle button.
This will be a great improvement and outstanding feature compared to Alexa, Google and  several other implemetations which use a toggle button to.

kakopappa, 2018-07-23:
@DonKracho just to be clear, you want a one button to turn on/off group of switches right ?

DonKracho, 2018-07-23:
Hi,
no, Groups of switches may be done with alexa. But goups and timers in your app are highly appreciated too. Right now I like to have two buttons for one switch. One ON button and one OFF button for one device ID.

The RF remote has On/Off buttuns to. This is important for the programming procedure of one RF outlet. If you plug it in to mains it starts blinking for about 20 seconds. Then it is in programming mode and you have to press ON to attach it to the desired remote channel. Pressing OFF then the outlet forgets about this channel. When pressing Master Off it forgets all programming. One outlet can be linked to up to 4 remotes

Due to this programming procedure it is important not to have deicated on and off buttons. No Smart app does provide these right now.

433 MHz RF outlets are somewhat special beause sometine they do not react due to interference in the band and the can not send back the real status. But the newer "digital" generation has got much more reliable and uses 32 round robin codes and can handle up to 15 switches, the last code pair will be master on/off. I had success controling them with an Arduino Uno scetch I developed some time ago. With a larger antenna attached to the Arduino TX I can control them all over my house.

If no one else needs this feature I may implement it as a local webserver page with separate buttons sending on off comands via a websocket to Sinric.

AshokBaijal, 2018-07-24:
Separate on off buttons will increase clutter on the mobile screen. My preference would be the present toggle button.

kakopappa, 2018-07-24:
i am thinking of putting a setting in the Menu -> Setting where you can specify which type you perfer.