kakopappa/sinric #94: Add French Langage in Alexa skill

stephcervetti, 2018-07-24:
Hi kakopappa,
Thank you so much for your API and all software already done. It's works fine but I m using Alexa in French langage so i would like to know if is it possible de translate to be compliant in French ? Maybe i can help you if you have a special file ? 
If you have any questions, i m available 
Have a good day
Regards

kakopappa, 2018-07-24:
Thanks.  

I have been thinking about adding French support for a while now and never got around to it. 

I will post the content which requires to be translated to French again and update this issue. 

congrats on the world cup

stephcervetti, 2018-07-24:
Thanks you so much.
So if you need help for translate or just to read your translation, i will
be available

Have a good day

*Stéphane CERVETTI*
Tél : 06 81 24 40 36

kakopappa, 2018-07-24:
One Sentence Description *
Connect your WeMos mini D1, ESP8266, ESP32 or Arduino development boards with Amazon Alexa for free!

Detailed Description *
With Sinric, you can connect your WeMos mini D1, ESP8266, ESP32, Arduino development boards with Alexa without emulating as a Belkin switch.

How to use?

1. Goto sinric.com

2. Create an account and log in

3. Create a new device. eg: tv

Use the API key displayed on the home screen to connect your development board.

Here is an example of how to connect :

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

replace the "Place your API key here", "your wifi ssid", "your wifi password" in the code with your details

4. Install / Enable Sinric smart home skill

5. Ask Alexa to discover devices.

6. Say Alexa, turn on "tv"

kakopappa, 2018-07-31:
@stephcervetti any luck?

nicovi21, 2018-10-10:
Hi @kakopappa 

One Sentence Description *
Connectez gratuitement votre WeMos mini D1, ESP8266, ESP32 ou vos cartes de développement Arduino avec Amazon Alexa ou Google Home !

Detailed Description *
Avec Sinric, vous pouvez connecter votre WeMos mini D1, ESP8266, ESP32 ou vos cartes de développement Arduino avec Alexa sans avoir à émuler un interrupteur Belkin.

Comment l’utiliser ?
1.	Allez sur Sinric.com
2.	Créez un compte et connectez-vous
3.	Ajouter un équipement. exemple: télé
Utilisez la clé API affichée sur l’écran d’accueil pour connecter votre carte de développement.
Voici un exemple indiquant comment se connecter :
https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino
Remplacez dans le code "Place your API key here", "your wifi ssid", "your wifi password" avec vos informations.
4.	Installez/Activez le skill Sinric smart home
5.	Demandez à Alexa de découvrir les appareils.
6.	Dites, Alexa allume la télé.

Example Phrases *
1.	Alexa, allume la télé
2.	Alexa, allume la lumière de la chambre
3.	Alexa, diminue la lumière de la chambre

Keywords:
Connecter, ESP8266, to, Alexa, connecter, Wemos, D1, mini, Alexa, Arduino, Alexa

kakopappa, 2018-10-11:
Thanks.

I am still waiting for the Spanish skill to be approved. Seems it takes long time for approval now. Once it's done I will submit for for French