kakopappa/sinric #95: Spanish translation help required

kakopappa, 2018-07-24:
Please help me to translate this to Spanish so I can publish

One Sentence Description *
Connect your WeMos mini D1, ESP8266, ESP32 or Arduino development boards with Amazon Alexa or Google Home for free!

Detailed Description *
With Sinric, you can connect your WeMos mini D1, ESP8266, ESP32, Arduino development boards with Alexa without emulating as a Belkin switch.

How to use?

Goto sinric.com

Create an account and log in

Create a new device. eg: tv

Use the API key displayed on the home screen to connect your development board.

Here is an example of how to connect :

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

replace the "Place your API key here", "your wifi ssid", "your wifi password" in the code with your details

Install / Enable Sinric smart home skill

Ask Alexa to discover devices.

Say Alexa, turn on "tv"

Example Phrases *

Alexa, turn on tv
Alexa, turn on bedroom lights
Alexa, dim bedroom lights
Keywords:
Connect, ESP8266, to, Alexa, connect, Wemos, D1, mini, Alexa, Arduino, Alexa

Andreslion, 2018-08-27:
Spanish translation
--------------------------------------------------------------------------------------------------------------
Conecte su Wemos mini D1, ESP8266, ESP32 o placa de desarrollo Arduino con Amazon Alexa o con Google Home gratis!

Con Sinric, ud puede conectar su Wemos mini D1, ESP8266, ESP32 o placa de desarrollo Arduino con Alexa sin tener que emular un interruptor Belkin.

Como usarlo?

Entre a Sinric.com

Cree una nueva cuenta y haga conexion.

Cree un nuevo dispositivo, por ejemplo: TV

Utilize la llave API mostrada en la pantalla de inicio para conectarse a su propia placa de desarrollo.

Aqui podra ver un ejemplo de como conectarse:

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

Remplace los campos "Place your API key here", "your wifi ssid", "your wifi password" en el código con sus propios detalles.

Instale / habilite el Skill Sinric smart home en su app Alexa.

Ordene a Alexa descubrir nuevos dispositivos. 

Diga "Alexa, enciende la TV"

Frases de ejemplo:
"Alexa, enciende la TV"
"Alexa, enciende las luces del cuarto"
"Alexa, tamiza las luces del cuarto"

Palabras clave:
Connect, ESP8266, to, Alexa, connect, Wemos, D1, mini, Alexa, Arduino, Alexa
--------------------------------------------------------------------------------------------------------------

There you go my friend! Thanks for giving us the AWESOME Sinric!

kakopappa, 2018-09-12:
hi, is this Spanish (ES) or Spanish (MX) ?

Andreslion, 2018-09-13:
Should work for both.

kakopappa, 2018-09-23:
OK I have finished Italian and now I have submitted the Spanish skill. I will let you know when there is an update.

soinel, 2019-03-30:
Hola kakopappa, una pregunta con Sinric es relativamente sencillo pero ¿se puede ver el estado, si esta encendido o apagado?.
por ejemplo yo utilizo un modulo ESP8286 ó ESP8285 y utilizo una entrada para un pulsador, con el objeto de poder utilizar el interruptor en modo manual. ¿que debo enviar a sinric para que se entere que ha cambiado de estado?

gracias
Saludos

kakopappa, 2019-04-01:
@soinel according to google translate you want to send switch state manually to server? check this
https://github.com/kakopappa/sinric/blob/master/arduino_examples/how_to_update_server_example.ino

send it when you change the status once. do not send it in a loop always and overload the server. This is how you get banned

soinel, 2019-04-14:
Thank you.
I had it in front of me and I did not see it.

        digitalWrite(relayPin1, HIGH);
        setPowerStateOnServer(DEVICE1, "ON");
        estado = 1;