kakopappa/sinric #96: Italian translation help required

kakopappa, 2018-07-24:
Please help me to translate this to Italian so I can publish

One Sentence Description *
Connect your WeMos mini D1, ESP8266, ESP32 or Arduino development boards with Amazon Alexa or Google Home for free!

Detailed Description *
With Sinric, you can connect your WeMos mini D1, ESP8266, ESP32, Arduino development boards with Alexa without emulating as a Belkin switch.

How to use?

Goto sinric.com

Create an account and log in

Create a new device. eg: tv

Use the API key displayed on the home screen to connect your development board.

Here is an example of how to connect :

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

replace the "Place your API key here", "your wifi ssid", "your wifi password" in the code with your details

Install / Enable Sinric smart home skill

Ask Alexa to discover devices.

Say Alexa, turn on "tv"

Example Phrases *

Alexa, turn on tv
Alexa, turn on bedroom lights
Alexa, dim bedroom lights
Keywords:
Connect, ESP8266, to, Alexa, connect, Wemos, D1, mini, Alexa, Arduino, Alexa

marcovolpato00, 2018-07-31:
Here you are:

One Sentence Description *
Connetti le schede di sviluppo WeMos mini D1, ESP8266, ESP32 o Arduino con Amazon Alexa gratuitamente!

Detailed Description *
Con Sinric, è possibile connettere le schede di sviluppo WeMos mini D1, ESP8266, ESP32 o Arduino con Alexa senza il bisogno di emulare un dispositivo Belkin.

Come si usa?

Vai su sinric.com

Crea un account ed accedi

Crea un nuovo dispositivo. es: tv

Usa la API key mostrata nella pagina principale per connettere la tua scheda di sviluppo.

Qui è disponibile un esempio su come connettersi :

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

sostituisci "Place your API key here", "your wifi ssid", "your wifi password" con i tuoi dati

Installa ed abilita la skill Sinric

Chiedi ad Alexa di trovare nuovi dispositivi

Di' "Alexa, accendi la 'tv'"

Example Phrases *

Alexa, accendi la tv
Alexa, accendi la luce della camera da letto
Alexa, attenua la luce della camera da letto

kakopappa, 2018-07-31:
Thanks a lot for your help. I have submitted for review.

kakopappa, 2018-09-02:
they took a long time to test and it failed. I will correct it and submit again 💯

kakopappa, 2018-09-20:
Thanks guys. skill is now live.

closed 2018-09-20 by kakopappa