kakopappa/sinric #97: Japanese translation help required

kakopappa, 2018-07-31:
Please help me to translate this into Japanese so I can publish

One Sentence Description *
Connect your WeMos mini D1, ESP8266, ESP32 or Arduino development boards with Amazon Alexa or Google Home for free!

Detailed Description *
With Sinric, you can connect your WeMos mini D1, ESP8266, ESP32, Arduino development boards with Alexa without emulating as a Belkin switch.

How to use?

Goto sinric.com

Create an account and log in

Create a new device. eg: tv

Use the API key displayed on the home screen to connect your development board.

Here is an example of how to connect :

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

replace the "Place your API key here", "your wifi ssid", "your wifi password" in the code with your details

Install / Enable Sinric smart home skill

Ask Alexa to discover devices.

Say Alexa, turn on "tv"

Example Phrases *

Alexa, turn on tv
Alexa, turn on bedroom lights
Alexa, dim bedroom lights

Keywords:
Connect, ESP8266, to, Alexa, connect, Wemos, D1, mini, Alexa, Arduino, Alexa

ryorobot, 2019-02-11:
## Japanese translation

One Sentence Description *
WeMos mini D1、ESP8266、ESP32、その他Arduino開発ボードをAmazon AlexaやGoogle Homeと無料で接続できます！

Detailed Description *
Sinricを使うと、Belkinのスマートプラグをエミュレートすることなく、WeMos mini D1、ESP 8266、ESP32、その他Arduino開発ボードをAlexaと接続できます。

使い方

sinric.comにアクセスします。

アカウントを作成してログインします。

新しいデバイスを作成します。例: テレビ

ホーム画面に表示されているAPIキーを使用して開発ボードを接続します。

接続方法の例:
https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino
コード内の"Place your API key here", "your wifi ssid", "your wifi password"を使用する環境に応じて書き換えます。

Sinricスマートホームスキルをインストールし有効化します。

Alexaに「デバイスを探して」と話します。

「アレクサ、"テレビ"をつけて」と話します。

Example Phrases *
アレクサ、テレビをつけて
アレクサ、寝室のライトをつけて
アレクサ、寝室のライトを薄暗くして

Keywords:
接続, ESP8266, to, Alexa, connect, Wemos, D1, mini, Alexa, Arduino, Alexa

---
Hello, I tried Sinric in the account of the Amazon US.
My ESP32 was able to work with Alexa, it's amazing!
Thank you for your good platform.
I am looking forward to using Sinric in Japanese!

kakopappa, 2019-02-12:
@ryorobot Thanks lot. 

I will submit for approval tonight

ryorobot, 2019-02-12:
Thank you for quick response!
I am looking forward to the skills being released.

kakopappa, 2019-03-14:
@ryorobot  Japanese skill is available now https://www.amazon.co.jp/dp/B078RGYWQQ

cheers!

closed 2019-03-14 by kakopappa

kakopappa, 2019-10-09:
@ryorobot Hi there. 

can you help me with Sinric Pro release as well please?

ryorobot, 2019-10-13:
Hi @kakopappa .
Sinric Pro? That sounds very interesting!
I would like to cooperate.

kakopappa, 2019-10-13:
Awesome, that is very kind of you.,

https://github.com/sinricpro/language-files/blob/master/ja/amazon_alexa_skill.txt

This is the language file for Amazon Alexa skill. I have ran it through Google translate. It does a pretty good job but sometimes it needs small corrections. Can you take a look and modify it where appropriate, please? The English file is here https://github.com/sinricpro/language-files/blob/master/en/amazon_alexa_skill.txt

Apart from that, I like to make pro fully Japanese compatible. Including the website, portal, and backend. I see there is a big Japanese DIY maker community and it will help them immensely if you could help me to translate the other language files as well. Of course, in return, I will add freebies for your account when we go live the end of this month.

Thanks a lot

ryorobot, 2019-11-02:
Hi @kakopappa .
I'm sorry for being late, I sent a pull request with an optimized Japanese description.
Please let me know if you have any other translations!

I also discovered that the sample sketch link in the description of Amazon US's Alexa skill is incorrect.

kakopappa, 2019-11-02:
@ryorobot Thanks for your help

Yes i do like your help with translating the portal and messages. they are in the same folder
https://github.com/sinricpro/language-files/tree/master/ja

About the US skill issue. can you be more specific, please?

ryorobot, 2019-11-02:
@kakopappa 
Thank you for telling me the folder of text to translate.

Although it is a trivial problem, sample sketch link of description of Alexa skill of Amazon US
https://github.com/sinricpro/esp8266-esp32-sdk/tree/master/examples/Switch
Is a 404 error.

kakopappa, 2019-11-02:
@ryorobot Thanks. I will check what's the best possible resolution for this.

what do you think about the Sinric V2? did you find it hard/easy to setup? overall do you have any suggestions to improve the process?