kakopappa/sinric #99: Thermostat Read Temperature

cwstewart63, 2018-08-03:
Hi,
For my project (home automation of my pool), I just need to read the pool temperature.
For the alexa Thermostat home skill I believe it is now supported via the alexa.TemperatureSensor interface skill. Your example in the arduino Thermostat.ino does not include a read temperature function.
Can you advise if the sinric interface does support read temperature (any example) or I should just pursue a custom alexa skill with say thingspeak.
Congrats on the wemo switch emulator. Works great for my pool pump / light controls.
Just need you to emulate a Nest Thermostat to have a complete solution without a cloud based device.

Deimos1994, 2018-08-25:
@cwstewart63, it supports, here is an example:
https://github.com/kakopappa/sinric/blob/master/arduino_examples/thermostat_example.ino

Just use this function to send temperature to Sinric:
```cpp
//eg: setTargetTemperatureOnServer("deviceid", "25.0", "CELSIUS")
void setTargetTemperatureOnServer(String deviceId, String value, String scale) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["action"] = "SetTargetTemperature";
  root["deviceId"] = deviceId;
  
  JsonObject& valueObj = root.createNestedObject("value");
  JsonObject& targetSetpoint = valueObj.createNestedObject("targetSetpoint");
  targetSetpoint["value"] = value;
  targetSetpoint["scale"] = scale;
   
  StreamString databuf;
  root.printTo(databuf);
  
  webSocket.sendTXT(databuf);
}
```
It works with both google home and alexa, I've just tried.

netamego, 2018-11-01:
Hi @Deimos1994. Hi have Google Home and i want to query temperature. I have a working nodemcu with ds18b20 and i want to try Sinric. Please can you share your full sketch?. I now i need to change wifi conf, api keys and deviceid but i am interested how and where pin definitons goes. Thabks a lot.

cwstewart63, 2018-11-01:
Alternatively you can use thingspeak. If you want just a temperature sensor then yes the sinric approach is easiest. I wanted some custom status and command values so ended up using thingspeak mqtt with a custom Alexa skill. Sample code attached.
Not sure what your issue with the ds18b20 is. There are plenty of examples online:
#include <DallasTemperature.h>    // Include one wire libraries
#include <OneWire.h>
#define ONE_WIRE_BUSA 12  // Temperature 1 connected to Pin 12
  // Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWireA(ONE_WIRE_BUSA);
  // Pass our oneWire reference to Dallas Temperature. 
DallasTemperature sensorsA(&oneWireA);

void loop() {
      // read Datawire type analog inputs - comment out if using Thermistor
 sensorsA.requestTemperatures();
 temperature = (int) (sensorsA.getTempCByIndex(0) );  //
}

[Temp_to_ThingSpeakEample.zip](https://github.com/kakopappa/sinric/files/2537136/Temp_to_ThingSpeakEample.zip)

netamego, 2018-11-01:
Woow. This project and its followers are actually amazing. Thank you very much guys!!!

Giuseppe-P, 2018-12-21:
Sorry I do not understand how to operate the thermostat sinric .... The thermostat for heating should have the logic of reading the temperature from a sensor and the activation of a relay based on the setpoint ... You have an example .... Then how do I add the widget on sinric to see the given temperature?

emnik, 2018-12-21:
you can only set the target temp and change it or send an on off command. The whole logic of the thermostat should be handled at the esp or a raspberry pi or ... that will be connected to temp sensors relays etc...

Giuseppe-P, 2018-12-21:
I do not understand what you mean.
in fact the logic must be on esp.
but I do not understand how the firmware is written.
I do not see any code concerning the temperature reading.
no logic to manage on / off automatically based on the setpoint set.
the thermostat should work with a logic on the node.
on the sirnic the thermostat is displayed as a switch.
it should not be this way, but it should have a reference temperature and a setpoint to set for heating.

I do not understand his sirnic logic.

emnik, 2018-12-21:
Let me put it this way.
Think that you implement a full thermostat using an esp. You connect temp sensors and relay on it, you read the temperature from the sensors and you let a user set a setpoint let's say via a webpage that is served through the esp. Then with the temp sensor data and the setpoint available you implement the thermostat logic. Also you need to have an on off button at the web page that the user can use to turn on or off the thermostat let's say when he leaves his home or returns to it. 
A classic thermostat would proudly have a tft screen and buttons so that the user would be able to perform the same actions as with the webpage described above.
Now think that Alexa/sinric is an alternative to the buttons-screen or to the webpage. The user can say "Alexa turn on/off thermostat" or "Alexa set thermostat to 20 degrees" and Alexa will send the appropriate command to sinric just like if you pressed a real button or a button on a webpage. Then you programmatically perform the requested action.
That being said, for the time being you should think Alexa/sinric as a very basic interface to a thermostat that you will build. 
If and when the whole API of Amazon Alexa thermostat is implemented then things might be more close to what you initially anticipated...

Giuseppe-P, 2018-12-21:
Thank you for your time. I created this http://souliss.net/articles/DST/ based on the souliss framework. I understood what you have described to me but I can not understand how to insert the relay in the code under which condition it will activate.

Giuseppe-P, 2018-12-22:
Ok, I understand what you mean ... But how do I know the temperature of the sensor via Alexa? Can you tell me where to insert the data in the thermostat example?

emnik, 2018-12-22:
You send the temperature to Alexa via the setTargetTemperatureOnServer function of the example. Remember you cannot send ambient temperature! You only send the setpoint! 
I cannot tell you where you should put this function in your code as I'm not familiar with the souliss framework and with a

Giuseppe-P, 2018-12-22:
I sent that link to let you know that I'm quite familiar with C ++ code ... That's my project. The thing that is not clear to me with sinric thermostat is how to know from the outside with Alexa the indoor ambient temperature .. Ok .. I can know the setpoint ... But how do I determine how many degrees I have to increase to activate the relay heating based on the logic written on esp? The sinric thermostat is very strange.

emnik, 2018-12-22:
Ok, as I had the same problem with you I'll tell you again that you have wrong in your mind what the sinric thermostat is at the moment. 
Forget that it is a thermostat!!!
It is not!
It is just a basic interface for your thermostat to trigger on off and set the setpoint. Nothing more.
What you ask me is a job that your thermostat will do and it is independant of sinric.
Let's break out down to what you can do:
1) You can ask Alexa to set a target temp via voice. Then you get the JSON response it sends and use the data to send them to your thermostat that should update its setpoint to the new one.
2) You can ask Alexa to turn on or off the thermostat and again take the JSON response check if the command was on or off and send it to your thermostat to turn on or off itself accordingly.
3) You can ask Alexa to increase or decrease the temperature (meaning again the target temperature) and get the Delta value from the JSON response. Then you send the Delta value to your thermostat and the thermostat should update its target point adding that Delta value to its current setpoint.
So you see that the idea is
Voice command Alexa -> JSON response -> send data from response to the thermostat -> handle the data in the thermostats code. Sinric has nothing to do with the last part.
Think about it a little...

Giuseppe-P, 2018-12-22:
Okok .. Perfetto ora mi è chiaro ... Grazie per la tua collaborazione ..... Volevo chiedere se c'è un widget in sinrico che invia dati temperatura da un sensore ...