kakopappa/sinric #100: setPowerStateOnServer(String deviceId, String value) does not update the server?

DonKracho, 2018-08-06:
When checking with Alexa or the Sinric App I can not see that the state really gets updated. I need this because my RF wall outlets can be controlled from another source too.

Am I doing something wrong with the fuction setPowerStateOnServer sending the deviceID as given for the device and "ON" or "OFF" as value?

Will it be possible to use this command as a heartbeat to set the online state of the device?  Even it the ESP device is not powered up Alexa thinks that the devices are online.

kakopappa, 2018-08-07:
Hello, sorry about the late response.

I have made a small change so the Amazon skill will pass the testing.
(Needs a device to be online always) after the skill passes I can fix this.
Till then it will be like this. I will update you when it’s done

nnc6, 2018-11-20:
I think I have the same problem. I'm controlling the switch by Alexa and by a normal button. The logic part is ok and everything is working fine but I'm not able to update the status on server. Do you have any news?

nnc6, 2018-11-20:
here's the code: 
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "d5ffb669-2ab1-4320-80c1-cbe1696e2f04" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "TIM-36312017" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "veRyYDXZeaXmstEi" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

const int relayPin = D1;
const int buttonPin = D2;
int relayStatus;
int buttonState=0;       // current state of the button
int lastButtonState;     // previous state of the button

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void setPowerStateOnServer(String deviceId, String value);
void setTargetTemperatureOnServer(String deviceId, String value, String scale);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "5bf4541beb220f3c039cf6c4") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    
     digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH
  } 
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);    
  }     
}

void turnOff(String deviceId) {
   if (deviceId == "5bf4541beb220f3c039cf6c4") // Device ID of first device
   {  
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     
     digitalWrite(relayPin, LOW);  // turn off relay with voltage LOW
   }
   else {
     Serial.print("Turn off for unknown device id: ");
     Serial.println(deviceId);    
  }
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch(type) {
    case WStype_DISCONNECTED:
      isConnected = false;    
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
      isConnected = true;
      Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
      Serial.printf("Waiting for commands from sinric.com ...\n");        
[... 153 lines cut ...]
// If you are going to use a push button to on/off the switch manually, use this function to update the status on the server
// so it will reflect on Alexa app.
// eg: setPowerStateOnServer("deviceid", "ON")

// Call ONLY If status changed. DO NOT CALL THIS IN loop() and overload the server. 

//eg: setPowerStateOnServer("deviceid", "CELSIUS", "25.0")

// Call ONLY If status changed. DO NOT CALL THIS IN loop() and overload the server. 
void setTargetTemperatureOnServer(String deviceId, String value, String scale) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["action"] = "SetTargetTemperature";
  root["deviceId"] = deviceId;
  
  JsonObject& valueObj = root.createNestedObject("value");
  JsonObject& targetSetpoint = valueObj.createNestedObject("targetSetpoint");
  targetSetpoint["value"] = value;
  targetSetpoint["scale"] = scale;
   
  StreamString databuf;
  root.printTo(databuf);
  
  webSocket.sendTXT(databuf);
}