kakopappa/sinric #104: Alexa can't link to sinric

jejetica, 2018-08-17:
Hello, 

I tried to link to sinric and I get always an error: unable to link skill, try later. I already tried different amazon accounts. maybe someone has an idea what it can to be?

kakopappa, 2018-08-17:
Yah .. Sorry about this.

 Server SSL certificate has expired I will renew it as soon as I finish
this beer (in few hours)

jejetica, 2018-08-17:
ah ok, 
I try tomorrow again. Enjoy your beer :)

kalingaa, 2018-08-17:
if you setup cloudflare in front, the SSL is free from them and the service is also free, plus you don't have to worry about local server cert renewal and it has many other benefits as well.

kakopappa, 2018-08-17:
Thanks machange . I will take a look this weekend

kakopappa, 2018-08-17:
[image]

done should be fine now. please check and let me know if otherwise

@kalingaa sometimes I sometimes I listen to talk katha. keep up the good work bro.

kalingaa, 2018-08-17:
@kakopappa honored to have you as a listener buddy! join us for an episode, even via Skype ?

kakopappa, 2018-08-20:
@kalingaa thanks. i will drop you a message when i am free.

jejetica, 2018-08-20:
sorry for the late answer, it works. Thank you!

RanTalbott, 2018-09-26:
I'm still seeing this, but only when Firefox "helpfully" remembers that I had previously logged in via alexa.sinric.com at some point. If I fix the URL and go to just plain sinric.com, I get a different webpage, and the login works correctly. Maybe whatever host is serving https://alexa.sinric.com/login is still using the old certificate?

shikoghaly, 2019-01-03:
Hi kakopappa pls renew the server

shikoghaly, 2019-01-03:
Hi kakopappa pls renew the server because now closed