kakopappa/sinric #105: Add Projector Screen

cpyarger, 2018-08-19:
Hello, Can we add a projector screen device? Essentially the same as a switch only accepts "up" and "down" instead of "on" and "off" as command words

kakopappa, 2018-08-20:
sure. I can put this in next release. Is this for Alexa or Google home ?

kakopappa, 2018-08-25:
I have added the projector type now

for Alexa, 
Alexa.PowerController
Alexa.PercentageController
Alexa.ChannelController

for Google
action.devices.types.OUTLET
action.devices.traits.OnOff

cpyarger, 2018-08-25:
Sweet, So now I can just call Alexa screen up
and the following snippet should then accept the command in an Arduino or
esp8266, correct?

if(action == "PowerController") {
           String value = json ["value"];
            if(value == "UP") {
                ScreenUp(deviceId);
            } else {
                ScreenDown(deviceId);
            }
}
}
Warmest Regards,

Christopher P. Yarger

Phone: 802-505-7574
Skype: cpyarger

/************************************
Based upon email reliability concerns,
please send an acknowledgment in response to this note.

cpyarger, 2018-08-25:
> AFK
>
> You have to check the serial monitor for the commands and handle channel
> controller properly
>
> https://developer.amazon.com/docs/device-apis/alexa-channelcontroller.html
>
> Alexa turn up/down <device name> will forward the Alexa request to your ESP
>
> Iet me know if you have a problem I will check when I get back home
>
> On Sat, 25 Aug 2018 at 12:05 PM Christopher P. Yarger <

I feel like I may have made a miscommunication, I am trying to control a
motorized screen like this..

https://www.amazon.com/Homegear-110-HD-Motorized-Projector/dp/B00J22TM7Y

by saying

alexa screen up
or
alexa screen down

in order to turn on the motor for the screen with a relay.

I believe the alexa skill Alexa.PowerController only responds to

alexa screen on
and
alexa screen off

like I currently have running with the switch device type

Warmest Regards,

Christopher P. Yarger

Phone: 802-505-7574
Skype: cpyarger

/************************************
Based upon email reliability concerns,
please send an acknowledgment in response to this note.

kakopappa, 2018-08-27:
Closest api I can find for you is this one
https://developer.amazon.com/docs/device-apis/alexa-channelcontroller.html

Look at SkipChannels. It supports voice commands like this

Alexa, channel up on device
Alexa, channel down on device

I think you can omit on here and say 
Alexa, channel up device

Try creating a projector device and ask Alexa to

Alexa, channel up on device
Alexa, channel down on device

Let me know how it goes.

clabnet, 2018-12-30:
Hi, please suggest me for using this command with Window Shutter.
Alexa reply me : "Sorry the device cannot reply", in italian "Mi dispiace, il dispositivo xxxxxx non risponde".
Because I have not reply any command.
Amazon developer site says: 
> "You must send a Response event if a request to change or skip channels was successful. The response should include a context object that reports the channel after the request completes."

How to reply a response event using your amazing software ?

Thank you, and happy new year.