kakopappa/sinric #106: Set lights to a color stopped working

Deimos1994, 2018-08-19:
It worked a couple days ago, but now when I say for example "set lights to Orange", Google home responds: **"Actually, the Main lights doesn't support that functionality."**

kakopappa, 2018-08-20:
well, i did not make any changes to the code except renew the  certificate. 

try recrearing the light, it drop me a message here I will take a look.

al8943, 2018-08-20:
I have the same problem, some time after the certificate update google started saying sinric is not available. I un-linked and re-linked to restore and afterward started getting the light doesn't support that functionality message to color change requests.
Creating a new device of the type light gives the same result.

al8943, 2018-08-20:
The color change still works on Alexa.

pico563, 2018-08-20:
same problem

pico563, 2018-08-20:
working OK now

Deimos1994, 2018-08-20:
@pico563 Just tried, it doesn't work for me, still saying "Actually, the Main lights doesn't support that functionality."
Maybe google something changes on their side.

pico563, 2018-08-20:
@Deimos1994 
yes you are correct, it stops again.
I will tell you what I did:
Settings--> Home control ->Manage accounts
1-unlink   [test] Sinric
2-add new    [test] Sinric
3-sync devices

go to 
 https://console.actions.google.com/u/0/project/sinric-4730c/simulatorcreate
and press test

now I tried it again but no luck

kakopappa, 2018-08-21:
hi guys,

sorry about the late response. so to test this I created a test device called "light" and this is the response I send back to Google when it requests devices. 
 

```

{  
   "requestId":"650641601724711203",
   "payload":{  
      "agentUserId":"zzzz",
      "devices":[  
         ...,
         {  
            "id":"xxxx",
            "type":"action.devices.types.LIGHT",
            "traits":[  
               "action.devices.traits.OnOff",
               "action.devices.traits.Brightness",
               "action.devices.traits.ColorTemperature",
               "action.devices.traits.ColorSpectrum"
            ],
            "name":{  
               "defaultNames":[  
                  "light"
               ],
               "name":"light",
               "nicknames":[  
                  "light"
               ]
            },
            "willReportState":false,
            "deviceInfo":{  
               "manufacturer":"SINRIC",
               "model":"SINRIC",
               "hwVersion":"1.0",
               "swVersion":"1.0"
            },
            "customData":{  
               "deviceId":"xxx"
            },
            "attributes":{  
               "colorModel":"RGB",
               "temperatureMinK":2000,
               "temperatureMaxK":6500
            }
         }
      ]
   }
}
```
This match with the docuemntation 
https://developers.google.com/actions/smarthome/guides/light#response

I think google changed something

kakopappa, 2018-08-21:
posted it here
https://plus.google.com/114858751617885172243/posts/MuqDw3Ukeq2

pico563, 2018-08-22:
GH now answer for any device
"It looks like Sinric is unavailable right now"
On the other hand 
Amazon echo works fine.

kakopappa, 2018-08-22:
@pico563 try again, https://google.sinric.com/login seems to up. may be api server restarted

pico563, 2018-08-22:
https://google.sinric.com/login 
Cannot GET /undefined

https://sinric.com/login
is OK

pico563, 2018-08-23:
[image]

My RGB strip still working with GH

kakopappa, 2018-08-23:
how about others?

pico563, 2018-08-23:
[image]

pico563, 2018-08-23:
Yesterday I got a Sinric  Error      "It looks like Sinric is unavailable right now"
to solve it I
1-unlink [test] Sinric
2-add new [test] Sinric
3-sync devices

after Sinric was OK

but still no color control

kakopappa, 2018-08-23:
I will put some more time on this weekend and try to figure out what’s
going on.

Deimos1994, 2018-08-23:
I rarely get "Sinric unavailable", maybe once a month, not more often.
@pico563 Well, interesting, seems like one of your lights: "Bed Room Strip" supports color, but other 2 lights don't support that... How is it possible? it could be possible if we could choose custom traits, but we can't choose what traits to use as of right now, there is no such option at Sinric.com

pico563, 2018-08-23:
@Deimos1994 
As you can see "Bed Room Strip" & "Living Room Strip" are both on Magic Home WiFi but other  2 lights are on Sinric.
All lights work fine with Amazon Echo Dot .
the only thing i like with GH is   "turn on/off my TV"  depending on the room you are sitting + searching & playing youtube , Netflix  videos with your voice. (Chromecast)

Deimos1994, 2018-08-23:
@pico563 ok, it explains everything, because I thought it was Sinric.
well... it means it is not Google problem, at least it is not completely their problem :) but I still think Google changed something, and that something caused some kind of incompatibility.
@kakopappa As a suggestion, you can try to remove color temperature traits:
`"action.devices.traits.ColorTemperature",`
 and not just because google guy pointing this out, but because there is some logic there.
I remember when color was working I still had an issue with the white color, all colors that I received was in RGB format, except white, the white color was received as color temperature in Kelvin.
Btw, Sinric would be much better if we could choose traits for each device... but it is probably not easy to implement, just an idea for the future... Anyways, thanks for your work, I'll wait as long as it takes :)

pico563, 2018-08-23:
even with white color temperature GH was OK
here is how I convert color temperature to RGB
```
//-------------------------------------------------------------------------
//Algorithm to Convert color Temperature (K) to RGB
//http://www.tannerhelland.com/4435/convert-temperature-rgb-algorithm-code/

  void K_to_RGB(float K, byte rgb[]){        // Kelvin to RGB

      float r,g,b;   
      K /= 100;  
/*---------------------Calculate Red-----------------------*/        
    if (K <= 66) { r = 255;      
      }else{  r = K - 60;
          r = 329.698727446 * pow(r,-0.1332047592);
              if (r < 0) { r = 0 ;}
              if (r > 255) { r = 255 ;}}
/*---------------------Calculate Green---------------------*/        
    if (K <= 66) { g = K;        
        g = (99.4708025861 * log(g) - 161.1195681661) ;
        if (g < 0) { g = 0 ;}
        if (g > 255) { g = 255 ;}
    }else{
        g = K - 60 ;
        g = 288.1221695283 * pow(g,-0.0755148492) ;
        if (g < 0) { g = 0 ;}
        if (g > 255) { g = 255 ;}}
/*---------------------Calculate Blue-----------------------*/        
    if (K >= 66) { b = 255;        
    }else if (K <= 19) {b = 0;
      }else{ b = K - 10;
            b = ( 138.5177312231 * log(b) - 305.0447927307 );
            if (b < 0) { b = 0 ;}
            if (b > 255) { b = 255;}}

        rgb[0] = round(r);
        rgb[1] = round(g);
        rgb[2] = round(b);

       rr = rgb[0] ; gg = rgb[1] ; bb = rgb[2] ;
    }        
    
/*==========================================================*/

```
**HERE is part of my sketch for GH that deal with white**

```
/*G--------------------------SHADES of WHITE---------------------------------*/
//http://www.vendian.org/mncharity/dir3/blackbody/UnstableURLs/bbr_color.html

        if (temperature == 3000 ){  // Warm White
          rr = 255 ; gg = 180 ; bb = 107 ;    // 255 180 107  #ffb46b
          RGB_Out(rr,gg,bb) ;}  
   else if (temperature == 3000 ){  // Soft White
          rr = 255 ; gg = 180 ; bb = 107 ;    // 255 180 107  #ffb46b
          RGB_Out(rr,gg,bb) ;}  
   else if (temperature == 4000 ){  // Cool White
          rr = 255 ; gg = 209 ; bb = 163 ;    // 255 209 163  #ffd1a3
          RGB_Out(rr,gg,bb) ;}  
   else if (temperature == 4100 ){  // Moonlight
          rr = 255 ; gg = 211 ; bb = 168 ;    // 255 211 168  #ffd3a8
          RGB_Out(rr,gg,bb) ;}  
   else if (temperature == 5000 ){  // White
          rr = 255 ; gg = 228 ; bb = 206 ;    // 255 228 206  #ffe4ce
          RGB_Out(rr,gg,bb) ;}  
   else if (temperature == 5000 ){  // DayLight
          rr = 255 ; gg = 228 ; bb = 206 ;    // 255 228 206  #ffe4ce
          RGB_Out(rr,gg,bb) ;}  
   else if (temperature == 6500 ){  // Snow
          rr = 255 ; gg = 249 ; bb = 253 ;    // 255 249 253  #fff9fd
          RGB_Out(rr,gg,bb) ;}  

        else{ RGB_Out(rr,gg,bb) ;}

    }
}

```

[https://github.com/kakopappa/sinric/issues/80#issuecomment-402074385](url)

Deimos1994, 2018-08-23:
@pico563, I agree. btw, nice code, it can be very helpful. But since I had an issue only with one color I don't even need to convert anything, I just can make all three colors 255 and that is it.
But, the cool way to do it would be to set what my lights can do in the first place... because some lights only support RGB (in theory RGB lights support everything...), but some lights only can change brightness but not the color, or only can change brightness and color temperature, or only can be on/off, that is why I think the ability to choose traits is a good idea.
I just think that maybe Google made impossible to use color temperature and RGB at the same time. I mean... nothing changed on Sinric, then why it stopped working? mystery :)

pico563, 2018-08-24:
**Here is how Amazon app control lights I hope soon we will find Sinric with a better  UI.**
[image]

[image]

pico563, 2018-08-24:
**Here is another app**

[[image]](url)

Fleker, 2018-08-24:
I am curious as to why the lights support both RGB and temperature colors at the same time. I don't believe that is related to this issue.

The errors appear to be a bit sporadic, making me wonder if there's some error happening in the Sinric service related to auth or handling requests.

kakopappa, 2018-08-25:
Thanks everyone for your time and helping me to fix this. After spending some time I found out the reason.

1. First removed temperature colors to check whether it works. nope, that did not work

2. I saw in https://developers.google.com/actions/smarthome/traits/colorspectrum#response
it says "colorModel": "rgb" in **bold** as  rgb in lower case then i remembered Sinric used to send {colorModel:"RGB"}

Changed it to send "colorModel": "rgb" and it worked!

I think someone at Google made a change a few days ago and there is a case-sensitive logic somewhere to check whether lights support rgb, which broke Sinric

1. Validator must be fixed. It should check for case 

2. Documentation should be updated
https://developers.google.com/actions/smarthome/guides/light#response

here is proof it's working
 
https://youtu.be/_rObK1rJ6lY