kakopappa/sinric #110: How can I make a server?

GiampaMan, 2018-08-31:
I Kakopappa!
First I want to tell you again that you are doing a great work! 
I want to tell you that I am doing a project on my school with air conditioners. So I need to add a lot of devices to controll each one by rutines on the Alexa app. So I am trying to make my own server (like Sinric) to connect it to my AWS account. 
Can you help me with that process?
I am a student of 15 years old. I have a YouTube channel of electronics tutorials in Spanish. If you want to visit this is the link:
https://www.youtube.com/channel/UCwq0rcuP3Z6iPP_VitoRShw

I dont know much about servers but I want to do my own Sinric server and my own skill (with other name).
So, in conclusion, I am starting in the IoT world and I want to do my own server.
Can you help me with that? 
If I tell you to do the server I can pay you for your work?
Remember I am starting in this.
Have a nice day!

kakopappa, 2018-08-31:
> How can I contact you by private?

aruna.tennakoon@gmail.com
support@irdevkit.com

debadityadas, 2019-06-30:
Sir even my aim is the same to open my own server same like sinric so Sir please tell me how can I contact you

sivateja369, 2020-01-04:
Hello 

I am SCIVTEJ,

I run orphanage home in my place and I’m having 100 to 500 switches which it takes hours to turn them off when all students leave the home so I need to build my own sinric like server and Alexa skill to donate them can you please help me on that

Thanks