kakopappa/sinric #111: Alexa Gerät hinzugefügt, keine Reaktion

MischaI, 2018-08-31:
[image]
Alexa hat Gerät gefunden. Kann auch in der AlexaApp ein/ausschalten. Auch SinricApp. Aber Relay wird nicht geschaltet. Auch im COM Port ändert sich nicht... Einige male Gerät deinstalliert, andere Wemosd1mini und Relayshields ausprobiert. Hilft nichts. Habe auch mit standart API-KEY und meinem von sinric.com getestet. Was soll ich machen?

kakopappa, 2018-08-31:
According to google translate you cannot get the relay to work? So create a new sketch and test the relay controller first.

```
 #include <Arduino.h>
int in1 = 7; // <-- Change to the correct GPIO number

void setup() {
  pinMode(in1, OUTPUT);
  digitalWrite(in1, HIGH);
}
void loop() {
  digitalWrite(in1, LOW);
  delay(3000);
  digitalWrite(in1, HIGH);
  delay(3000);
}
```

Once you get the relay controller to work

Use this sketch https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

Change the API key, WiFi username. password and move the code inside this sketch.

According to google translate you cannot get the relay to work? So create a new sketch and test the relay controller first.

```
#include <Arduino.h>
int in1 = 7; // <-- Change to the correct GPIO number

void setup() {
  pinMode(in1, OUTPUT);
  digitalWrite(in1, HIGH);
}
void loop() {
  digitalWrite(in1, LOW);
  delay(3000);
  digitalWrite(in1, HIGH);
  delay(3000);
}
```

Once you get the relay controller to work

Use this sketch https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

Change the API key, WiFi username. password and move the code inside this sketch.

MischaI, 2018-09-01:
Relay work witch test sketch. But witch this sketch(https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino) no. I set my SSID, PWD, API KEY...
[image] I cant see im sketch pin for relay(D1(pin5)). Should i manuell input this pinNr?

MischaI, 2018-09-01:
SInric Sketch works good. i just restart my router, reload sketch, alexa find a new device(but as belkin socket). it works fine
[image]

kakopappa, 2018-09-01:
Ok i think you are confused here or I do not understand your problem

Alexa support two methods

1. UDP broadcasts on LAN.
2. Skill.

1. UDP broadcasts on LAN.

based on my previous project:
https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch 

This works based on UDP. This is nothing to do with Sinric project. When you ask Alexa to discover new devices, it will send a UDP broadcast to find the devices on the network and show. When you ask Alexa to turn on/off it will send a UDP broacast

2. Skill.

You have to install the Skill and Create an account in Sinric and configure to use the API key.
sketch(https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino) 

I have not added the relay on/off code in about sketch

if you want on/off relay code. use this

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example_wemos_d1_mini_with_relay.ino

Hope this answers your question

kakopappa, 2018-09-01:
if you look at the serial output, you can see it says "turn on unknown device". That means you did not handle the code for your device id.

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example_wemos_d1_mini_with_relay.ino#L29

changed to your device id here

MischaI, 2018-09-01:
So wie es aussieht, hätte ich Probleme mit udp. Weil nachdem ich Router neugestartet habe, den Sketch noch mal geuploadet, alexa hat relay gefunden, und jetzt funktioniert alles per alexa. Skill ist deaktiviert.
Gestern hat bei mir alexa mit skill relay gefunden. Ich konnte in alexa und der app relay an/aus schalten, aber relay hatte keine Reaktion. Im com Port ide hat auch nichts gezeigt. Relay an sich mit anderem Code funktioniert. Jetzt nach dem ich Router neugestartet habe funktioniert alles mit alexa gut. Ich glaube, man kann thread zumachen.

closed 2018-12-18 by MischaI