kakopappa/sinric #112: Alexa says 'OK' even if device is not connected

malicmt, 2018-09-01:
Hi bro,
Thank you for your great work....
Before, when I was using your WeMos switch - Alexa used to say "<device name> is not responding" if the ESP was disconnected from network or power.
But, now when I use Sinric - Alexa is acting like the ESP is online even if it is not. This is making it difficult for me to trace out some kind of issues.

Is this how it works - or am I missing something?

saikhurana, 2018-09-07:
yep this is how it works. even if esp disconnects from the server, the server will respond yes.

closed 2018-09-08 by malicmt

gizmocuz, 2018-12-01:
This is not good. You could check if the esp is connected to the websocket with the API key
If not, return false to the Amazon query

Is it possible to implement this? Now we do not know what is causing it.. the switch.. the esp?