kakopappa/sinric #113: you got it, setting the reception to 26.4 degree

pico563, 2018-09-05:
@kakopappa 
Hi
on my Living Room GH mini  if I say
1-command  ------>       Set reception to 26.4 degree.
[WSc] get text: 
 {"deviceId":"xxxx","action":"action.devices.commands.SetTemperature","value":{"temperature":26.5}}

2-command  ------>        Set thermostat to 26.4 degree  (name of thermostat is reception)
 [WSc] get text: 
{"deviceId":"xxxx","action":"action.devices.commands.ThermostatTemperatureSetpoint","value":{"thermostatTemperatureSetpoint":26.39999961853027}}

for both commands GH answer  "**you got it, setting the reception to 26.4 degree**"
is that correct ?

pico563, 2018-09-05:
[image]

pico563, 2018-09-05:
[image]

kakopappa, 2018-09-06:
Yes that’s seems correct. What’s the problem?

pico563, 2018-09-06:
The first command is for light bulb color temperature in degree Kelvin .( Reception lights)
{"deviceId":"xxxx","action":"action.devices.commands.SetTemperature","value":{"temperature":26.5}}

The minimum color temperature I know is 1000 K so **what is 26.5 ?**

[image]

at Device STATES: if I ask **" What is my current light color temperature?"** ---> no answer
https://developers.google.com/actions/smarthome/traits/colortemperature

-----------------------------------------------------------------------------------------------------
The second command is for thermostat temperature in degree  Celsius . ( Reception thermostat){"deviceId":"xxxx","action":"action.devices.commands.ThermostatTemperatureSetpoint","value":{"thermostatTemperatureSetpoint":26.39999961853027}}

google is not differentiating  between the two commands.
at the beginning I thought that both are for thermostat degree .

--------------------------------------------------------------------------------------------------------
please add fan speed control to Sinric so I can use it in my A/C project

https://developers.google.com/actions/smarthome/traits/fanspeed

Thanks

Deimos1994, 2018-09-06:
@pico563 Hi, this is what you should get for the color temperature (if you use lights): `'{"deviceId":"5b7ad84eeXXXXXXXX4b8d2","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"white","temperature":5000}}}'`
For the thermostat I always get this:
`'{"deviceId":"5b2e6d6bXXXXXXX42a8f9","action":"action.devices.commands.ThermostatTemperatureSetpoint","value":{"thermostatTemperatureSetpoint":26.39999961853027}}'
`

But you've said that you getting this: `{"deviceId":"xxxx","action":"action.devices.commands.SetTemperature","value":{"temperature":26.5}}` I've never received this payload, I've tried setting temperature with name, or just saying thermostat, but nothing. Are you sure that you getting this payload? bacause it is strange...

pico563, 2018-09-06:
@Deimos1994 
yes, you are correct. for me also color temperature  is OK
```
{"deviceId":"xxxx","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"cool white","temperature":4000}}}
 {"deviceId":"xxxx","action":"action.devices.commands.BrightnessAbsolute","value":{"brightness":8}}

```
I just tried it again for 26.6 degrees
here is the result.

```
[WSc] get text: {"deviceId":"5b8xxxxf946","action":"action.devices.commands.ThermostatTemperatureSetpoint","value":{"thermostatTemperatureSetpoint":26.60000038146973}}
Sensor OK. NO Warnings
26.12       26.60       °C
26.20       57.50       DHT
[WSc] get text: {"deviceId":"5b8xxxxf946","action":"action.devices.commands.SetTemperature","value":{"temperature":26.5}}

```

pico563, 2018-09-06:
I really don't know .
now sure, it is not color temperature so what it is ?

pico563, 2018-09-06:
I will add both on my sketch 

```
         if (action == "action.devices.commands.ThermostatTemperatureSetpoint"  || "action.devices.commands.SetTemperature" ) {
            deg =( json["value"]["thermostatTemperatureSetpoint"] ||  json["value"]["temperature"] );

```
I hope it will work .
thanks

pico563, 2018-09-06:
@Deimos1994 
`If I understood correctly action.devices.commands.SetTemperature is a trait for devices other than thermostats... you should not receive that.
`

that is correct.  I don't know how it passes through Sinric.

`I think you can try delete the device and then creating it again.`

I will.

Deimos1994, 2018-09-06:
@pico563 Every device has its traits. Thermostat usually has `action.devices.traits.TemperatureSetting` but in theory it also can have `action.devices.traits.TemperatureControl` and if so then you will probably start to receive `action.devices.commands.SetTemperature` but I don't know how and why it happens.

closed 2018-09-11 by pico563