kakopappa/sinric #115: TV skill: "Sorry, TV isn't responding"

gustavolyra, 2018-09-12:
When i ask alexa to change channel, ESP receive the command but alexa say "Sorry, TV isn't responding".
This only happens when i ask to change channel, change volume works fine.

Do ESP need to send something to alexa after receive the command to change channel?

kakopappa, 2018-09-12:
Sinric sends success back to Alexa and forward the command to esp. normally
you don’t have to do anything

kakopappa, 2018-09-17:
> yes, i see the command
> WiFi connected. IP address: 192.168.100.22
> [WSc] Service connected to sinric.com at url: /
> Waiting for commands from sinric.com ...
> [WSc] get text:
> {"deviceId":"5b8df18bf332cf2dc424f99f","action":"AdjustVolume","value":{"volume":10,"volumeDefault":true}}
> [WSc] get text:
> {"deviceId":"5b8df18bf332cf2dc424f99f","action":"AdjustVolume","value":{"volume":-10,"volumeDefault":true}}
> [WSc] get text:
> {"deviceId":"5b8df18bf332cf2dc424f99f","action":"setPowerState","value":"ON"}
> [WSc] get text:
> {"deviceId":"5b8df18bf332cf2dc424f99f","action":"ChangeChannel","value":{"channel":{},"channelMetadata":{"name":"tnt"}}}
>
> ESP receive all commands, only the ChangeChannel alexa say: "Sorry, TV
> isn't responding"
>
> My code:
> `#include <Arduino.h>
> #include <ESP8266WiFi.h>
> #include <ESP8266WiFiMulti.h>
> #include <WebSocketsClient.h> //
> https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
> #include <ArduinoJson.h> //
> https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
> #include <StreamString.h>
> [... 188 lines cut ...]
> StreamString databuf;
> root.printTo(databuf);
>
> webSocket.sendTXT(databuf);
> }`

I checked the code for ChangeChannel . Sinric does send the correct
response back to Amazon. Issue is likely from Amazon side

gustavolyra, 2018-09-17:
is there anything else i could do, or just wait until amazon solve the problem ?

victorantonio21, 2018-09-18:
> is there anything else i could do, or just wait until amazon solve the problem ?

I believe you need to report the issue back to amazon so they can try to solve it, on your alexa app go to feedback option

Comotio82, 2018-11-05:
Hello, I have same problem. Was someone able to fix it?

jdecmac, 2018-11-17:
Hello,
I have the same problem. How and what should we tell Amazon to fix it?

Thank you,

regards

eddcaton, 2018-11-22:
Same issue here. Still not working

amitabh-srivastav, 2018-11-24:
Code is working fine as far as Volume, Mute, Unmute, On/Off are concerned with proper reply from Alexa. However, any attempt to change channel gives a response "Sorry, Tata Sky/TV is not responding". 
I think you need to re-look on the issue.

amitabh-srivastav, 2018-11-24:
The response on serial port is as follows:
[WSc] get text: {"deviceId":"5bddf1832d8b221d153d7d57","action":"setPowerState","value":"OFF"}
[WSc] get text: {"deviceId":"5bf83accded8b075eb77ef7f","action":"SetMute","value":{"mute":false}}
[WSc] get text: {"deviceId":"5bf83accded8b075eb77ef7f","action":"SetMute","value":{"mute":true}}
[WSc] get text: {"deviceId":"5bf83accded8b075eb77ef7f","action":"SkipChannels","value":{"channelCount":1}}
[WSc] get text: {"deviceId":"5bf83accded8b075eb77ef7f","action":"ChangeChannel","value":{"channel":{"number":"147"},"channelMetadata":{}}}
[WSc] channel Number: 147

kakopappa, 2018-11-27:
here is the problem for "ChangeChannel, Amazon docs does not say anything. 
https://developer.amazon.com/docs/device-apis/alexa-channelcontroller.html#changechannel 

so I send the same response for "SkipChannels" call. May be it does not match with what Alexa expect it to be. Need to resolve this with Amazon Alexa

kakopappa, 2018-11-27:
raised an issue here
https://forums.developer.amazon.com/questions/192048/whats-the-appropriate-response-for-changechannel.html

amitabh-srivastav, 2018-11-28:
By the way, did you check the following, mentioned on Alexa docs.

Response
You must send an Response event if a request to change or skip channels was successful. The response should include a context object that reports the channel after the request completes.

Response Example

{
  "context": {
    "properties": [
      {
        "namespace": "Alexa.ChannelController",
        "name": "channel",
        "value": {
          "number": "1234",
          "callSign": "callsign1",
          "affiliateCallSign": "callsign2"
        },
        "timeOfSample": "2017-02-03T16:20:50.52Z",
        "uncertaintyInMilliseconds": 0
      }
    ]
  },
  "event": {
    "header": {
      "messageId": "abc-123-def-456",
      "correlationToken": "dFMb0z+PgpgdDmluhJ1LddFvSqZ/jCc8ptlAKulUj90jSqg==",
      "namespace": "Alexa",
      "name": "Response",
      "payloadVersion": "3"
    },
    "endpoint":{
      "endpointId":"appliance-001"
    },
    "payload":{ }
 }
}
ErrorResponse
You should reply with an error if you cannot complete the customer request for some reason. See Alexa.ErrorResponse for more details.

Additional Sample Code
See the sample request and response messages in the Alexa smart home GitHub repo:

ChannelController

kakopappa, 2018-11-28:
Thanks. Last time I check this is the response for the command I send back.
I will check again  on this weekend

amitabh-srivastav, 2018-11-29:
Thanks, you may have a look at this one :
https://github.com/alexa/alexa-smarthome/tree/master/sample_messages/ChannelController

amitabh-srivastav, 2018-12-05:
Dear @kakapappa sir, any update on the issue?

Comotio82, 2018-12-12:
Dear @kakopappa, could you investigate this issue again, please?

kakopappa, 2018-12-12:
@Comotio82 @amitabh-srivastav  sorry guys. i will check it on this weekend again

minimatte77, 2018-12-13:
hi, even I myself problem with Italian language, however, compliments for the work done

amitabh-srivastav, 2018-12-20:
@kakopappa Any update on the issue as promised.

kakopappa, 2018-12-28:
found the issue and fixed it. enjoy!

Comotio82, 2018-12-28:
I can confirm it works well now. Thank you! Enjoy your beer.

kakopappa, 2018-12-28:
thanks. sorry about the delay. I was sick like 2 weeks.

amitabh-srivastav, 2018-12-29:
Thanks a lot dear @kakopappa !

clabnet, 2018-12-30:
Today the same issue.
https://github.com/kakopappa/sinric/issues/105#issuecomment-450570380
How to solve ?

kakopappa, 2018-12-31:
@clabnet   server will send OK back to Alexa once it forwards the command to your ESP module, You do not have to do anything for this. 

can you check again, please ? I can turn on/off fine.

SparkyDan555, 2019-09-06:
I'm still getting "Sorry the device is not responding" when setting inputs / changing channels on a TV device? I have disabled and re-enabled the Alexa skill but still alexa says not responding even though the ESP8266 recieves the command.

MacSass, 2020-01-15:
> 
> 
> I'm still getting "Sorry the device is not responding" when setting inputs / changing channels on a TV device? I have disabled and re-enabled the Alexa skill but still alexa says not responding even though the ESP8266 recieves the command.

Hi @kakopappa,
I can confirm that for TV the input controller - although it is working ok - gets an Alexa response of "Sorry the device is not responding".

In my skill I do receive a good request from my point of view:
[WSc] get text: {"deviceId":"xyz","action":"SelectInput","value":{"input":"HDMI 1"}}

So I think something with the response you are sending back to Amazon must be wrong, so Alexa thinks it was not working ...

In June 2018 you mentioned you had checked your response and it looked ok ... but I doubt that Amazon has this bug for so long, as many people would be using input controller by now.

Would you be able to double check your response again?

If I find the time to convert my .ino to sinric.pro, I will do so in order to see if this is the same on sinric.pro, but right now I do not have enough skill to change all to sinric.pro asap ...

Thanks for your help - MacSass