kakopappa/sinric #116: Got code working, but keeps rebooting

victorantonio21, 2018-09-18:
I'm using nodemcu Esp8266, uploaded the multi-wifi version of the code, after somedays I managed to upload the code successfully but the board keeps rebooting itself. At serial, after connecting to the ESP and setting up the WiFi this is what I got:

```
ets Jan  8 2013,rst cause:4, boot mode:(3,2)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
*WM: SETTINGS ERASED 
mounting FS...
reading config file
opened config file...
{"Networks":[{"SSID":"INT","Password":"111111111"}
parsed json
*WM: AutoConnect 
*WM: Connecting as wifi client... 
*WM: No saved credentials, skipping wifi 
*WM: Connection result: WL_NO_SSID_AVAIL
*WM: connectTimeout not set, ESP waitForConnectResult... 
*WM: Scan for known APs 
*WM: Connecting to:  
*WM: INT 
*WM: connectTimeout not set, ESP waitForConnectResult... 
*WM: Connection result:  
*WM: 3 
*WM: AutoConnect: SUCCESS 
*WM: STA IP Address: 192.168.0.101

connected to: INT
local ip: 192.168.0.101
```

and then reboots again, and again.... 

I appreciate every help

kakopappa, 2018-09-19:
have you tried updating the multi-wifi library to the latest version?

victorantonio21, 2018-09-30:
Yes, I did, but got no luck. 
But I've managed to get the other code working on booth alexas V1 and V2 by mixing the modified code with the original updated one. Thank you for everything!!