kakopappa/sinric #117: Alexa can't discover devices

lackdaz, 2018-09-19:
Hey @kakopappa, thanks for being so patient with everyone here. I've read virtually most of your issue threads, but I still have a small issue with the alexa app integration bit.

I'm familiar with arduino and I've got all the code bits to work with your sinric API (tested on your console and there's output on my serial console).

But on the alexa app bit, I have linked your sinric smart home skill and I can discover the devices if I manually click 'Discover', but, Alexa is unable to discover the devices if I try the voice command route, ie 'Alexa, discover devices'. But the most important bit, I'm unable to use any of the 'voice invocations' to trigger the api call to sinric.

I've tried re-linking the skill, forgetting the devices and rinse repeat but to no avail. You think the problem might be on the amazon end?

I'm using desktop alexa app (mobile is not available in my country and I own an alexa)

kakopappa, 2018-09-19:
Thanks. 

Sorry your question is not clear. 

When you link the skill and ask Alexa to discover devices, can Alexa discover devices in your account ?

When you perform a basic command like on/off (via Alexa not the app), do you see the on/off command in the serial output?

You said you are using desktop alexa app. I have not tried this. I live in Thailand. When i open the iOS Alexa app and navigates to Smart Home Devices, I cannot see any smart home devices as well. (may be because not released in Thailand as well).  I contacted the supprot and they asked me to unlink the skill and try again / restart the router ect.. I gave up and made the Sinric android app

lackdaz, 2018-09-19:
I hope this montage explains the situation clearly.

I'm trying to communicate with Alexia via voice to turn on/off the switch, ie "Alexa, turn on switch 1", and not using a toggle switch on a mobile app -- which I think defeats the point of having alexa because there are simpler ways to implement remote switching via mobile (e.g. mqtt).

What am I doing wrong here? How does the sinric android app come in?

Thanks again for the fast response

kakopappa, 2018-09-19:
Thanks

this is strange. If you say "Alexa, turn on Eve" it should work.  It should send a command to ESP module (Make sure API Key is correct)

what about creating something like TV. "Alexa, turn on TV" ? Does it work? or still same

have you tried using Alexa ios/android app? if it doesnt work you should try contacting Amazon support.

lackdaz, 2018-09-19:
Alright I found the issue. The answer was buried in this thread in your earlier implementation:
https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/issues/22#issuecomment-353685276

I share an echo dot with my partner and we have separate accounts for the dot -- the conflict was that the echo was logged into her account while I was configuring the dot from my account on the desktop app -- hence none of the voice commands worked while the desktop UI could discover devices.

TLDR; I f**cked up and didn't check the Echo was currently on my account of my partner's. Do caution folks that if they are running individual accounts for a single Echo, they might have to set up sinric separately.

closed 2018-09-19 by lackdaz

Dilanravindu, 2018-12-16:
hello kakopappa ,amazon alexa android app ekath ekka use kanna puluwanda  .mama try kala but wada kale ne , devise add kana tanata gihin nawatinawa