kakopappa/sinric #119: More than 2 devices on ESP8266

malvar0929, 2018-09-25:
Great Job with this service. I got it running in very little time. The only issue I'm having is for a third device. I'm able to turn off and on 1 time only then no response however device 1 and 2 continue to work fine. I think my code for the third device may be incorrect. I'm using this with Alexa on a ESP8266.
Please see code below and I will appreciate any help you can give me. Thank you.

```
void turnOn(String deviceId) {

  if (deviceId == "xxxxxxxxxxxxxxxxxxxxxxxxxxxx") // Device ID of first device
  {
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    pinMode(charge, HIGH); //charge ROV
  }
  else if (deviceId == "xxxxxxxxxxxxxxxxxxxxxxxxxxxx") // Device ID of second device
  {
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(PiROV, LOW); //Turn-on ROV
  }
  else if (deviceId == "xxxxxxxxxxxxxxxxxxxxxxxxxxxx") // Device ID of third device
  {
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(Oled, LOW); //Turn-on display
  }
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);
  }
}
```

malvar0929, 2018-09-25:
`void turnOn(String deviceId) {

  if (deviceId == "5aa4de1d7ca6b616f79bd320") // Device ID of first device
  {
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    pinMode(charge, HIGH); //charge ROV
  }
  else if (deviceId == "5a9b55bab950f610a7e66c09") // Device ID of second device
  {
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(PiROV, LOW); //Turn-on ROV
  }
  else if (deviceId == "5ba9a4d59e1da81c77b12892") // Device ID of third device
  {
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(Oled, LOW); //Turn-on display
  }
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);
  }
}

void turnOff(String deviceId) {
  if (deviceId == "5aa4de1d7ca6b616f79bd320") // Device ID of first device
  {
    Serial.print("Turn off Device ID: ");
    Serial.println(deviceId);
    pinMode(charge, LOW);  // Turnoff charging ROV
  }
  else if (deviceId == "5a9b55bab950f610a7e66c09") // Device ID of second device
  {
    Serial.print("Turn off Device ID: ");
    Serial.println(deviceId);
    digitalWrite(PiROV, LOW); //Turnoff ROV
  }
  else if (deviceId == "5ba9a4d59e1da81c77b12892") // Device ID of third device
  {
    Serial.print("Turn off Device ID: ");
    Serial.println(deviceId);
    pinMode(Oled, LOW); //Turnoff display
  }
  else {
    Serial.print("Turn off for unknown device id: ");
    Serial.println(deviceId);
  }
}
`

kakopappa, 2018-09-25:
Does it print something in the serial when you turn on/off third device?

malvar0929, 2018-09-25:
Yes it works as normal in serial print, I have tested it multiple times.

kakopappa, 2018-09-25:
"The only issue I'm having is for a third device. I'm able to turn off and on 1 time only then no response however device 1 and 2 continue to work fine." Not quite sure what you mean here.

1. You have created 3 devices in Sinric. Lets say they are A, B, C

2. When you ask alexa to turn on off C, it does not work is it?

Do you see the request from the server printed on Serial ?

malvar0929, 2018-09-25:
When I ask to turn off C, I see the request from server printed on serial but my C does not turn off. If I reboot ESP8266 it works once (off/on) then it's ignored after first request, however I continue to see request from server in serial but does not trigger C.

kakopappa, 2018-09-26:
Have you tried connecting C to another GPIO?

DonKracho, 2018-10-10:
There is a german idiom: I can't see the forrest because there are so many trees:) It's just a copy paste error in the turnOff function for the 3rd device. Excange pinMode(Oled, LOW); by digitalWrite(Oled, LOW);

malvar0929, 2018-10-11:
I can't believe I didn't see that, however when you pointed that out, additionally I also noticed that I was setting the pin LOW for both OFF and ON operations. 
Perfect fitting idiom by the way. 
Thank you very much, all is working well now.

kakopappa, 2018-10-11:
damm.  i missed it too

closed 2018-10-11 by malvar0929