kakopappa/sinric #120: NodeMCU reboot loop

victorantonio21, 2018-09-25:
I have a NodeMCU esp8266 with 5 relays on it, and after using the Multi WIFI version, the device keeps reboting itself everytime after connecting to wifi network. The reason for it seems to be some watchdog crash but i dont know how to fix it. :/

kakopappa, 2018-09-25:
1. Make sure you have a good power supply.

2. Do you have loops? Likely because you have not called yield();
https://www.sigmdel.ca/michel/program/esp8266/arduino/watchdogs_en.html

victorantonio21, 2018-09-30:
I'm using a great power supply, The problem is in the Yield(); for sure, I'm gonna work on that latter for my new esp that I ordered. Once Again, Thank You so Much For Everything!