kakopappa/sinric #124: How to add TV skills

6575bala, 2018-10-04:
How to add tv skills like volume up/down and change channel sketch code.

kakopappa, 2018-10-04:
Create a TV device

When you ask Alexa to vol up/down, Alexa send a command to Sinric, and Sinric will forward those commands to your ESP module. 

Check the suppported Controllers from TV here
https://github.com/kakopappa/sinric/wiki/Supported-devices-and-their-capabilities.

You have to add the code to handle it in the MCU

Check the supported languages for these commands. Not all the languages are supported.

6575bala, 2018-10-04:
I am not software engineer so how to make the code in MCU please send copy of code. (exact parse)

kakopappa, 2018-10-04:
Do you know how to flash the code using Arduino to ESP?

kakopappa, 2018-10-05:
When you mute or unmute it will send a command. This line will print it in the serial console.

Serial.printf("[WSc] get text: %s\n", payload);

if you do not know. Look at youtube and learn how to use Arduino IDE to check serial messages.

Then add a condition like this

else if(action == "ChangeChannel") {

} else if(action == "xxxxxxxx") {

}

6575bala, 2018-10-05:
Sir i can't understand add a condition like this so(attach my sketch code) please add my sketch code like channel change and channel up/down and vol up/down.  

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> 
#include <ArduinoJson.h>
// IR part-------------------------------------------------------------
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>

#define IR_LED 4  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(IR_LED);  // Set the GPIO to be used to sending the message.

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "your wifi ssid name" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "your wifi password" // TODO: Change to your Wifi network password

#define API_ENDPOINT "http://sinric.com"
#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void TogglePower(String deviceId) {
  if (deviceId == "xxxxxxxxxxxxxxxxxxxxxx") // TODO: Change to your device Id for TV
  {  
     irsend.sendSAMSUNG(0xE0E040BF,32);  // Send a power signal to Samsung TV.
     delay(500);
  } 
  
  if (deviceId == "yyyyyyyyyyyyyyyyyyyyyyy") // TODO: Change to your device Id for Set Top Box
  {  
      
     irsend.sendRC6(0xC0000C,24); // Send a power signal to tatasky.
     delay(500);
  } 
}

void ToggleMute(String deviceId) {
  if (deviceId == "xxxxxxxxxxxxxxxxxxxxxxxxxx") // TODO: Change to your device Id for TV
  {  
     irsend.sendSAMSUNG(0xE0E0F00F,32);  // Send a power signal to Samsung TV.
     delay(500);
  } 
  
  if (deviceId == "yyyyyyyyyyyyyyyyyyyyyyyyy") // TODO: Change to your device Id for Set Top Box
  {  
      
     irsend.sendRC6(0xC0000D,24);  
     delay(500);
  }  

  }

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch(type) {
    case WStype_DISCONNECTED:
      isConnected = false;    
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
      isConnected = true;
      Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
      Serial.printf("Waiting for commands from sinric.com ...\n");        
      }
      break;
    case WStype_TEXT: {
        Serial.printf("[WSc] get text: %s\n", payload);
           
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject((char*)payload); 
        String deviceId = json ["deviceId"];     
        String action = json ["action"];
        
        if(action == "setPowerState") { 
            String value = json ["value"];
            if(value == "ON" || value == "OFF" ) {
                TogglePower(deviceId);
                }
        }

        else if(action == "ChangeChannel") {
          String ChannelName=json ["value"]["channelMetadata"]["name"];
          String ChannelNumber=json ["value"]["channel"]["number"];
          

          if(ChannelName=="national geographic"){
            irsend.sendRC6(0xC00007,24);  // Send IR code for remote button 7
           delay(500);
           irsend.sendRC6(0xC00000,24); // Send IR code for remote button 0
           delay(500);
           irsend.sendRC6(0xC00008,24); //Send IR code for remote button 8
           delay(500);
            Serial.println("[WSc] channel: " + ChannelName);
           }
          if(ChannelName=="star movies"){ 
           irsend.sendRC6(0xC00003,24);  // Send IR code for remote button 3
           delay(500);
           irsend.sendRC6(0xC00005,24); // Send IR code for remote button 5
           delay(500);
           irsend.sendRC6(0xC00003,24); // Send IR code for remote button 3
           delay(500);
            Serial.println("[WSc] channel: " + ChannelName);
          }
         }

       
        else if (action == "SetMute") {
          
         bool MuteAction=json ["value"]["mute"];
          if(MuteAction==true || MuteAction==false){ 
             ToggleMute(deviceId);
           }
          
        }

        else if (action == "test") {
            Serial.println("[WSc] received test command from sinric.com");
        }
        
      }
      break;
    case WStype_BIN:
      Serial.printf("[WSc] get binary length: %u\n", length);
      break;
  }
}

void setup() {
  Serial.begin(115200);
    irsend.begin();

 WiFiMulti.addAP(MySSID, MyWifiPassword);
  Serial.println();
  Serial.print("Connecting to Wifi: ");
  Serial.println(MySSID);  

  // Waiting for Wifi connect
  while(WiFiMulti.run() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  if(WiFiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.print("WiFi connected. ");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

6575bala, 2018-10-05:
sir i'll be waiting for your reply.

kakopappa, 2018-10-07:
I have added a new tv example
https://github.com/kakopappa/sinric/blob/master/arduino_examples/tv_example.ino

You have to change your sketch to send the IR signal accordingly. 

Send the IR signals in a separate sketch and make sure it works first and then integrate it to this sketch

6575bala, 2018-10-08:
sir doesn't supported this sketch.

6575bala, 2018-10-09:
sir send IR signal is working but how to integrate tv example sketch and IR signal sketch.

kakopappa, 2018-10-09:
copy the sending ir code inside the sketch

6575bala, 2018-10-09:
sir my IR code attach which place IR code add  tv example sketch.

#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>

#define IR_LED 4  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(IR_LED);  // Set the GPIO to be used to sending the message.

#if SEND_SAMSUNG
  Serial.println("SAMSUNG");
  irsend.sendSAMSUNG(0xE0E040BF, 32); // Sends power on/off signal to TV
#endif  // SEND_SAMSUNG
  delay(500);
#if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D9D02F, 32, 2); // Sends power on/off signal to Set Top box
#endif  // SEND_NEC
  delay(500);
#if SEND_SAMSUNG
  Serial.println("SAMSUNG");
  irsend.sendSAMSUNG(0xE0EOFOOF, 32); // Sends mute/unmute signal to Tv
#endif  // SEND_SAMSUNG
  delay(500);
#if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D99E817, 32, 2); // Sends mute/unmute signal to Set Top box
#endif  // SEND_NEC
  delay(500);
#if SEND_SAMSUNG
  Serial.println("SAMSUNG");
  irsend.sendSAMSUNG(0xE0E0E01F, 32); // Sends volume up signal to Tv
  #endif  // SEND_SAMSUNG
  delay(500);
#if SEND_SAMSUNG
  Serial.println("SAMSUNG");
  irsend.sendSAMSUNG(0xE0E0D02F, 32); // Sends volume down signal to Tv
  #endif  // SEND_SAMSUNG
  delay(500);
#if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D948B7, 32, 2); // Sends volume up signal to Set Top box
#endif  // SEND_NEC
 delay(500);
#if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D958A7, 32, 2); // Sends volume down signal to Set Top box
#endif  // SEND_NEC
 delay(500);
 #if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D9708F, 32, 2); // Sends channel up signal to Set Top box
#endif  // SEND_NEC
 delay(500);
 #if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D950AF, 32, 2); // Sends channel down signal to Set Top box
#endif  // SEND_NEC
 delay(500);
 #if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D930CF, 32, 2); // Sends Button 0  signal to Set Top box
#endif  // SEND_NEC
 delay(500);
 #if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D9D827, 32, 2); // Sends Button 1 signal to Set Top box
#endif  // SEND_NEC
 delay(500);
 #if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D9F00F, 32, 2); // Sends Button 2 signal to Set Top box
#endif  // SEND_NEC
 delay(500);
 #if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D9C03F, 32, 2); // Sends Button 3 signal to Set Top box
#endif  // SEND_NEC
 delay(500);
 #if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D99867, 32, 2); // Sends Button 4 signal to Set Top box
#endif  // SEND_NEC
 delay(500);
 #if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D98877, 32, 2); // Sends Button 5 signal to Set Top box
#endif  // SEND_NEC
 delay(500);
 #if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D9807F, 32, 2); // Sends Button 6 signal to Set Top box
#endif  // SEND_NEC
 delay(500); 
#if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D9906F, 32, 2); // Sends Button 7 signal to Set Top box
#endif  // SEND_NEC
 delay(500);
 #if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D9B847, 32, 2); // Sends Button 8 signal to Set Top box
#endif  // SEND_NEC
 delay(500); 
#if SEND_NEC
  Serial.println("NEC");
  irsend.sendNEC(0x22D9B04F, 32, 2); // Sends Button 9 signal to Set Top box
#endif  // SEND_NEC
 delay(500); 
 
}

kakopappa, 2018-10-09:
....
```
void turnOn(String deviceId) {
  if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of first device
  {  
   Serial.println("Turn on SAMSUNG");
   irsend.sendSAMSUNG(0xE0E040BF, 32); // Sends power on/off signal to TV
  }     
}
```

what happens when you do this?

6575bala, 2018-10-09:
sir error "ir send" was not declared in this scope.
[image]

6575bala, 2018-10-13:
sir volume up/down and channel up/down same time working how to solve  problem. sir please send example code.

       }else if(action == "AdjustVolume"){
           String value=json ["value"]["volume 10"]["volumeDefault"];
           
           if("Volume==10"){
           irsend.sendSAMSUNG(0xE0E0E01F,32); // Sends volume up signal to Tv
           delay(500); 
           irsend.sendSAMSUNG(0xE0E0E01F,32); // Sends volume up signal to Tv
           delay(500); 
           irsend.sendSAMSUNG(0xE0E0E01F,32); // Sends volume up signal to Tv
           delay(500);  
           irsend.sendSAMSUNG(0xE0E0E01F,32); // Sends volume up signal to Tv
           delay(500); 
           irsend.sendSAMSUNG(0xE0E0E01F,32); // Sends volume up signal to Tv
           delay(500); 
          } 
                  
        }else if(action == "AdjustVolume"){
          String value=json ["value"]["volume -10"]["volumeDefault"];

          if("Volume==-10"){ 
            irsend.sendNEC(0x22D958A7,32); // Sends volume down signal to Set Top box
           delay(500);   
           irsend.sendNEC(0x22D958A7,32); // Sends volume down signal to Set Top box
           delay(500);   
           irsend.sendNEC(0x22D958A7,32); // Sends volume down signal to Set Top box
           delay(500); 
           irsend.sendNEC(0x22D958A7,32); // Sends volume down signal to Set Top box
           delay(500);
           irsend.sendNEC(0x22D958A7,32); // Sends volume down signal to Set Top box
           delay(500);  
         }
        }

kakopappa, 2018-10-29:
code seesm to be correct. whats the problem here?

6575bala, 2018-10-29:
if(ChannelNumber=="353"){    357,370,375....
if(ChannelName=="national geographic"){    starmovies, starsports,....

6575bala, 2018-10-29:
Sir how to add this multiple ChannelName/ChannelNumber.

if(ChannelNumber=="353"){       357,370,375....
if(ChannelName=="national geographic"){        starmovies, starsports,....

kakopappa, 2018-10-29:
` if(ChannelName=="national geographic") {    
 }
 else if(ChannelName=="starmovies") {    
 }
 else if(ChannelNumber=="353") {
 }`

kakopappa, 2018-10-29:
just add it to the if condition.

 if(ChannelName=="national geographic") {    
...
 }
 else if(ChannelName=="starmovies") {    
...
 }
 else if(ChannelNumber=="353") {
....
 }

amitabh-srivastav, 2018-11-26:
Dear 6575bala
 For multiple channels you may use following code, used by me. Although, it still will say "Sorry, TataSky/ TV is not responding", but code works.

         else if (ChannelNumber.toInt() > 99){ 
          Serial.println("[WSc] channel Number: " + ChannelNumber) ;
          
          for (int count = 0; count <  ChannelNumber.length();count++){                
            switch (ChannelNumber.substring(count,count+1).toInt()){
              case 1:
                irsend.sendRC6(0xC00001,24);  // Send IR code for remote button 1
                Serial.println("1");
                break;
              case 2:
                irsend.sendRC6(0xC00002,24);  // Send IR code for remote button 2
                Serial.println("2");
                break;
              case 3:
                irsend.sendRC6(0xC00003,24);  // Send IR code for remote button 3
                Serial.println("3");
                break;
              case 4:
                irsend.sendRC6(0xC00004,24);  // Send IR code for remote button 4
                Serial.println("4");
                break;
              case 5:
                irsend.sendRC6(0xC00005,24);  // Send IR code for remote button 5
                Serial.println("5");
                break;
              case 6:
                irsend.sendRC6(0xC00006,24);  // Send IR code for remote button 6
                Serial.println("6");
                break;
              case 7:
                irsend.sendRC6(0xC00007,24);  // Send IR code for remote button 7
                Serial.println("7");
                break;
              case 8:
                irsend.sendRC6(0xC00008,24);  // Send IR code for remote button 8
                Serial.println("8");
                break;
              case 9:
                irsend.sendRC6(0xC00009,24);  // Send IR code for remote button 9
                Serial.println("9");
                break;
              case 0:
                irsend.sendRC6(0xC00000,24);  // Send IR code for remote button 0
                Serial.println("0");
                break;
            }
            delay(500);
          }
         }

6575bala, 2018-11-27:
Thanks for sharing the code details  AMITABH. how to change (Channel Name) code details.

6575bala, 2018-11-27:
> Dear 6575bala
> For multiple channels you may use following code, used by me. Although, it still will say "Sorry, TataSky/ TV is not responding", but code works.
>
> ```
>      else if [...]

Thanks for sharing the code details AMITABH. how to change (Channel Name) code details.

 @6575bala

Joshthynne, 2021-02-17:
A good, easy to follow tutorial for combining all of your IR remotes into one simple board, which integrates into Alexa via SinricPro. https://youtu.be/nh5adggIJYo