kakopappa/sinric #126: Webservice disconnected from sinric.com!

aososam, 2018-10-05:
My devices were working perfectly last night, today I cannot connect them to sinric.com, is it just me?

kakopappa, 2018-10-05:
Hi, today morning I moved the server inside cloudflair due to people sending large payloads and trying to hack the server.

aososam, 2018-10-05:
Hi! yes I can visit sinric.com and hit the test buttons of all my devices.
but all of them are now showing ¨Webservice disconnected from sinric.com!¨ I am using one NodeMCU and two ESP8266 Generic modules, I also updated the libraries and Boards but the problem persists

kakopappa, 2018-10-05:
can you try to connect using ip address?

noobipaul, 2018-10-05:
Hi, I am also seeing the same issue this morning. I am using wemos D1 mini esp8266 modules. Have tried re-flashing 2 modules a few times but still the same issue. Webservice disconnected from sinric.com

kakopappa, 2018-10-05:
looks like a DNS issue. It may take few mins to reflect. Can you try connecting using IP address please

aososam, 2018-10-05:
yes I will try that as soon as I can the IP is 104.24.111.26 right?

noobipaul, 2018-10-05:
I have just tried IP, is this correct --> webSocket.begin("104.24.111.26", 80, "/");
If so the problem gives me the same message Webservice disconnected from sinric.com

noobipaul, 2018-10-05:
Now I am connecting.

closed 2018-10-10 by kakopappa