kakopappa/sinric #129: [Feature request] Custom actions for the API?

lackdaz, 2018-10-06:
While this is not critically urgent, It'd be great if there is some way we could have more custom actions to the on/off/test

Like in my case, I have light bulbs that switches on/off to 3 different scene settings (https://www.philips.com.sg/c-m-li/choose-a-bulb/sceneswitch). I've gotten around this by creating a new device on the API that points to the same relay on my wemos -- a little hackey but works. 

Obviously this would change the UI and backend a bit - let me know if I can help with anything! I'm not entirely familiar with how alexa integrates with your smart skill.

kakopappa, 2018-10-07:
Hi,

I have been working on creating custom devices with custom actions for some time now. It appears to be a very difficult task. 

I will update this issue when it's ready.

lackdaz, 2018-10-09:
Thanks man good to hear!