kakopappa/sinric #130: Cannot link account

ElementProHD, 2018-10-06:
I keep on trying to connect my Alexa account and it won’t let me as it says failed to connect try again later

kakopappa, 2018-10-07:
I assume you are referring to linking the Skill here

Can you access this page?
https://alexa.sinric.com/login

ElementProHD, 2018-10-07:
Yes I can access this, what happens is a login and then receive the error message.

kakopappa, 2018-10-07:
Can you send a snapshot

ElementProHD, 2018-10-07:
“Unable to link your account”

ElementProHD, 2018-10-07:
https://imgur.com/a/vDSXxF6

kakopappa, 2018-10-07:
Sorry buddy I am not sure how to help you. I just tried linking and it works

[image]

Make sure your account is in one of the supported countries.

ElementProHD, 2018-10-07:
Do you have any idea what to try ,, I made another Sinric account and tired it and same error? I’m using a iPhone do you think that could be the issue

ElementProHD, 2018-10-07:
I have just tired on another device and still no luck

ElementProHD, 2018-10-07:
It’s england and I have had it working about 6 weeks ago but not it suddenly doesn’t want to work, it was working fine before and responding perfectly on a previous project and I plugged that back in and was also disconnected from Alexa.

kakopappa, 2018-10-07:
Not sure what I can do from this end. I think you have to contact support. You can do that from app.

kakopappa, 2018-10-28:
Can you check now, please? I made few changes

ElementProHD, 2018-10-28:
Hello I have resolved the issue it turns out that there is a bug on the Alexa App itself as I logged into several different accounts and had no luck linking them I then proceeded to logging in on my PC with 2 accounts and both successfully linked. The Alexa App was tired on IPhone X IOS 10 and IPhone 7 IOS 11