kakopappa/sinric #133: Arduino Uno R3 with EthernetShield W5100 error with WebSocketsClient.h

Comotio82, 2018-10-07:
Dear kakopappa,

first of all congratulations for your genius project and many thanks for your effort.
I tried some of your examples but get an error during compilation.
 
Arduino\libraries\arduinoWebSockets-master\src/WebSockets.h:37:2:` error: #error Version 2.x.x currently does not support Arduino with AVR since there is no support for std namespace of c++.
 #error Version 2.x.x currently does not support Arduino with AVR since there is no support for std namespace of c++.
Arduino\libraries\arduinoWebSockets-master\src/WebSockets.h:38:2: error: #error Use Version 1.x.x. (ATmega branch)

I checked with Version1 and Version2 of the libary.
Could you give me an advice what to do now?

Thanks beforehand.
Phil

kakopappa, 2018-10-07:
Oh, I have tried this only on ESP devices. Looks like your problem is
WebSockets
lib I used is not supported in AVR architecture.

Have you tried connecting using any other library?

kakopappa, 2018-10-07:
have you tried 1.3? Note in https://www.arduinolibraries.info/libraries/web-sockets says 
use 2.x.x for ESP and 1.3 for AVR

http://downloads.arduino.cc/libraries/github.com/Links2004/WebSockets-1.3.0.zip

Comotio82, 2018-10-07:
With WebSockets-1.3.0.zip it's possible to compile. Now I will try to change your example code without Wifi. Thank you!

frappesc, 2018-11-06:
there is any news? i'm trying to run it without an ESP but only with arduino and enc28j60. can someone help me?

ksb85, 2021-01-24:
@Comotio82 Did you get this working? If so, please can you share your example code, as I am now getting the following error:

`switch_example_2:148:3: error: 'webSocket' does not name a type
   webSocket.begin("iot.sinric.com", 80, "/");
   ^~~~~~~~~
switch_example_2:151:3: error: 'webSocket' does not name a type
   webSocket.onEvent(webSocketEvent);
   ^~~~~~~~~
switch_example_2:152:3: error: 'webSocket' does not name a type
   webSocket.setAuthorization("apikey", MyApiKey);
   ^~~~~~~~~
switch_example_2:155:3: error: 'webSocket' does not name a type
   webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
   ^~~~~~~~~
switch_example_2:156:1: error: expected declaration before '}' token
 }
 ^
exit status 1
'webSocket' does not name a type
`

I am trying to use the switch_example2.ino by @kakopappa, but I get the above error even when using Websockets 1.3