kakopappa/sinric #136: Setting a light back to white

kingkongsy, 2018-10-15:
Hi, 

after I set a light to a color (e.g. Red) I can´t find a way to reset it back to plain white.

I can select some prefab-colors via app and voice (like red, green, blue and turquoise).

I can also control the color temperature via the Alexa-App (not via voice-control, but this seems to be a known problem on Amazons side).

But after having selected e.g. red and then pressing on let´s say cold white, the color stays red, because "cold white" sets a different parameter (red -> SetColor vs. White -> SetColorTemperature). Is there a way to add a clean "white" to SetColor (HSL: 0/0/1)?

Best, K

kakopappa, 2018-10-19:
hi, thanks for reporting this issue.

this seems to be a bug in Sinric. It's not sending the color back to Amazon server when it requests the color status. I will upload the fix soon. 

Thanks

kakopappa, 2018-11-05:
Should be fixed now. Can you confirm please?

userfortyone, 2019-04-01:
It is not fixed.