kakopappa/sinric #139: Unable to link skill

MichaelM18, 2018-10-26:
Hi kakopappa,

Thank you very much for your service.

Unfortunately I'm running into a problem when linking the skill in the Alexa app on my iPhone. The error message says: "Unable to link your Skill. Please try again later." Please let me know if there's anything that I can do to resolve that issue. I already tried logging in with a second account but without success.

Any help would be greatly appreciated

Thanks,
Michael

[image]

kakopappa, 2018-10-28:
sorry about the late response. 

are you connecting from UK?

kakopappa, 2018-10-28:
Can you check now, please? I made few changes

ElementProHD, 2018-10-28:
Hello please try logging in to your Alexa account on a PC , (Not using Safari or the Alexa app itself) , I ran into the identical issue!!! I used google chrome on windows 10 pc then Sinric finllay paired with my account !!! Let me kno it this works

kakopappa, 2018-11-26:
I found an issue when linking using the Alexa app (iPhone) and fixed it last night.  (needs to handle an additional request when using phone). Alexa account on a PC works fine.

scheng123, 2018-11-29:
Are you sure you fixed it? Because I still have the issue now. At least on the iPhone Alexa App. The online version (https://alexa.amazon.com) works perfectly however.
Btw, keep up the superb work!

kakopappa, 2018-11-29:
@scheng123  thanks for letting me know. I will check again to make sure.

jrcrebollos, 2018-12-01:
HI Kakopappa,
Thank you for your wonderful service. I'm from Philippines and trying to connect my alexa to sinric. can you help me? i'm not able to link my alexa.

Thank you
JR

kakopappa, 2018-12-01:
I think Phil is not supported yet.

Have you tried linking using PC ? Goto alexa.amazon.com

jrcrebollos, 2018-12-01:
Hi Kakopappa,
I was able to link alexa. i downloaded the amazon alexa apk (not available in the philippines. searched in the google). in that app, i was able to link to sinric. in my pc, i cannot link my alexa to sinric. link unsuccessful.  hehe.
thank you!
Regards,
JR

kakopappa, 2018-12-01:
I just tried it on my pc 

[image]

Just went to https://alexa.amazon.com on my PC and can link the skill.  Alexa App has issues.

Do you have more than 1 aleca profiles ?

scheng123, 2018-12-01:
Try diagnosing the problem using [Web Inspector](url
https://developer.apple.com/safari/tools/) and compare the network requests with Chrome Dev Tools. That might get you somewhere.