kakopappa/sinric #140: Skill of Sinric for Spanish market?

jdecmac, 2018-10-27:
Hello,
I really like your Sinric project and in the coming days, Amazon will launch the Amazon echo with Alexa in Spain.

Are you planning to launch the Sinric Skill for Alexa in the Spanish market?

I am doing some projects with ESP8266 and I would love to be able to control them with Alexa through Sinric.

Thank you very much for your work

regards

kakopappa, 2018-10-27:
Thanks. 

Already. waiting for approval from Amazon.

cronos123, 2018-11-19:
No news?

kakopappa, 2018-11-19:
good question. I will take a look what happend.

kakopappa, 2018-11-19:
Close to two months now.. So I sent them a message. let's see their response.

jesusado, 2018-11-28:
Hello
I am also interested in your project Sinric Skill for Alexa for the Spanish market.
Still not available in Spanish?

Thank you very much for your work

regards

kakopappa, 2018-11-28:
Thanks, it’s in the review process. Should complete in few days

kakopappa, 2018-12-20:
Skill is available now.
https://www.amazon.es/dp/B078RGYWQQ

closed 2018-12-20 by kakopappa