kakopappa/sinric #141: update parameters via API

perlokan, 2018-10-29:
Hi,
can I change a parameter of a device via API?
For example the temperature of a thermostat.
Can you give me a curl example?
Thanks

kakopappa, 2018-10-30:
Right now it's allowed through only websockets (may be you know this already). 

can you tell me why do you want to do this? are you using Pi ?

perlokan, 2018-11-02:
Yes I use PI.

kakopappa, 2018-11-05:
I will update the API in the next milestone and update the API spec and sorry about the delay. 

I am spending lot of time these days on irdevkit.com with the new module testing.

perlokan, 2018-11-08:
Thank you.
It would be important for me to have also a timestamp for each device when it receives an Alexa command. Do you think you can add it?

greno, 2018-11-26:
I would also like to have the ability to send the current temperature to the server for my thermostat device.  Right now it looks like I can only send the target temperature, rather than the current temperature.  Using Arduino.

BTW, the API is working great for me. Very easy to set up and use.  Thanks!

kakopappa, 2018-12-01:
added to 1.4 release. check the api documentation for examples.

closed 2018-12-01 by kakopappa