kakopappa/sinric #142: Any chance to add Spanish for Google Home?

netamego, 2018-11-02:
Hi,

I can't believe the work you're doing, kakopappa. Moreover when you share your knowledge for free. Thanks alot for your work and your great shared knowledge philosophy.

Google Home it's supported in spanish. Are there any chance to add the Spanish language to Sinric?.

Thanks a lot.

kakopappa, 2018-11-05:
Thanks

Spanish language skill for Alexa is in progress. I have completed the setup and sent a test request to Amazon. They take ages to approve now.

netamego, 2018-11-05:
Thanks a lot kakopappa but i am talking about Google Home not Alexa. Thanks.

netamego, 2018-11-06:
I am using your sketch https://github.com/kakopappa/sinric/blob/master/arduino_examples/thermostat_example.ino and works perfect in spanish also. I ask "¿Cual es la temperatura interior?" and Google Home responds "La temperatura actual es de xxº"
Sorry for the issue. Theres is not issue here.

closed 2018-11-06 by netamego