kakopappa/sinric #145: Control RGB stripe?

drdrhase, 2018-11-04:
Hi, I found your awesome code and managed it to get working for my ESP8266. I connected 3 MOSFET to the ports D5, D6, and D7 to power 12V to the RGB channels of an RGB Stripe.
I used the "switch_example_wemos_d1_mini_with_relay.ino" sketch and defined the relay Pin given in this sketch to control ONE colour. But how can I connect the other Pins? Or am I doing it wrong? 

When I would use the "light_example.ino" example, which Pin would be activated?  ...confused...  ...need help...

kakopappa, 2018-11-05:
Hi,

Thanks. I think you are looking at the wrong example.  you should use the light example2 here
https://github.com/kakopappa/sinric/blob/master/arduino_examples/light_example2.ino

1. You can turn on/off using on off commands.

2. You have to handle the set color using 
https://github.com/kakopappa/sinric/blob/master/arduino_examples/light_example2.ino#L125

and change the RGB Stripe color accordinly

drdrhase, 2018-11-05:
Hi, thanks for your reply.
The sketch you suggest was the first I have tried, but I did not understood where I could define the pin numbers of the esp8266, where the RGB colors are phisicaly connected?
Sorry I am just starting...

kakopappa, 2018-11-05:
Well, in setup() you should set the pin modes and in the setColor you can change the LED colors.

Check this issue
https://github.com/kakopappa/sinric/issues/40

kakopappa, 2018-11-06:
here is an example code. I wrote this in note pad. may be there are compile errors

`
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h>
#include <ArduinoJson.h>
#include <StreamString.h>

#define API_KEY "" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define SSID_NAME "" // TODO: Change to your Wifi network SSID
#define WIFI_PASSWORD "" // TODO: Change to your Wifi network password
#define SERVER_URL "iot.sinric.com"
#define SERVER_PORT 80 

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void setPowerStateOnServer(String deviceId, String value);
void setTargetTemperatureOnServer(String deviceId, String value, String scale);

#define PIN_RED 5
#define PIN_GREEN 6
#define PIN_BLUE 3

void setup() {
  Serial.begin(115200);
  
  pinMode(PIN_RED, OUTPUT);
  pinMode(PIN_GREEN, OUTPUT);
  pinMode(PIN_BLUE, OUTPUT);
  
  WiFiMulti.addAP(SSID_NAME, WIFI_PASSWORD);
  Serial.println();
  Serial.print("Connecting to Wifi: ");
  Serial.println(SSID_NAME);  

  // Waiting for Wifi connect
  while(WiFiMulti.run() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  if(WiFiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.print("WiFi connected. ");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }

  // server address, port and URL
  webSocket.begin(SERVER_URL, SERVER_PORT, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", API_KEY);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
[... 168 lines cut ...]
            case 4:
                red = cc;
                green = aa;
                blue = _brightness;
                break;
            case 5:
                red = _brightness;
                green = aa;
                blue = bb;
                break;
            default:
                red = 0.0;
                green = 0.0;
                blue = 0.0;
                break;
        }
    }

    long ired = red * 255.0;
    long igreen = green * 255.0;
    long iblue = blue * 255.0;
    
    return long((ired << 16) | (igreen << 8) | (iblue));
}
`

drdrhase, 2018-11-06:
Man, you are completely awesome! Thanks allot!

I will test the code asap and give feedback to you.

drdrhase, 2018-11-06:
Hi kakopappa, 

there is a compile error in this section:

```
String hue = json ["value"]["hue"];
String saturation = json ["value"]["saturation"];
String brightness = json ["value"]["brightness"];

Serial.println("[WSc] hue: " + hue);
Serial.println("[WSc] saturation: " + saturation);
Serial.println("[WSc] brightness: " + brightness);
    
long color = HSBtoRGB(atof(hue), atof(saturation), atof(brightness)); 
    
 // Get the red, blue and green parts from generated color
int red = color >> 16 & 255;
int green = color >> 8 & 255;
int blue = color & 255;

setColor(red, green, blue);

    }
```

For `  long color = HSBtoRGB(atof(hue), atof(saturation), atof(brightness)); `

I get: 

> cannot convert 'String' to 'const char*' for argument '1' to 'double atof(const char*)'

kakopappa, 2018-11-14:
did you fix it? I am sorry I missed this completely.

flamboo, 2019-01-18:
The problem your having is along the same lines im having, I was wondering if either of you have been able to get it working?

drdrhase, 2019-01-18:
No I didn't.
But I installed tasmota firmware, which worked for me.

ivan978, 2019-03-31:
btw how to fix it the error

guicfreitas, 2019-12-12:
anyone knows how to fiz this problem?

sivar2311, 2019-12-12:
These are float or double values...not strings. Try it with this code:

```
    double hue = json["value"]["hue"];
    double sat = json["value"]["saturation"];
    double bri = json["value"]["brightness"];

```

The only thing you have to do is to convert it from HSV to RGB. 
(I never did this, because i am using fastLED library which is able to convert from HSV to RGB directly)

guicfreitas, 2019-12-12:
> These are float or double values...not strings. Try it with this code:
>
> ```
>     double hue = json["value"]["hue"];
>     double sat = json["value"]["saturation"];
>     double bri = [...]

Oh thank you so much but i solve the problem just using toCharArray()

sivar2311, 2019-12-12:
How do you do calculations with a char array? 
It compiles...yes..but you cant do calculations using a char array. 
Char arrays are stinrgs like "Hello" ...
You have to use integers, floats, doubles etc for calculations.

guicfreitas, 2019-12-12:
> How do you do calculations with a char array?
> It compiles...yes..but you cant do calculations using a char array.
> Char arrays are stinrgs like "Hello" ...
> You have to use integers, floats, doubles etc for calculations.

I’ve used the same code posted earlier, than i used toCharArray and the atof() like the parameter of the function ask for

sivar2311, 2019-12-12:
Okay, its the same like writing a letter on your pc, print it on your printer, scanning it as a jpg and send this jpg via emal...sure this is working too ;)

sivar2311, 2019-12-12:
What iam trying to say: Json data contains real double values. Why do you convert these into strings / char arrays and back to double or float values?

guicfreitas, 2019-12-12:
> What iam trying to say: Json data contains real double values. Why do you convert these into strings / char arrays and back to double or float values?

Sorry man I didn’t know about this my knowledge in json is really poor

sivar2311, 2019-12-12:
I don't want to be rude.
This isn't about json, this is about fundamental c / c++

guicfreitas, 2019-12-12:
> I don't want to be rude.
> This isn't about json, this is about fundamental c / c++

I dont want to be rude to but the creator of the code was not me, so how I didn’t know than json values return double values i’ve try to solve the problem that way, if knows this information from the beginning it was obvious that i dont have to convert strint to a char array than after converting to float!

sivar2311, 2019-12-12:
Ok  👍

sivar2311, 2019-12-12:
Have a look at [SinricPro](https://sinric.pro/). 
We have a Arduino library which is available via Arduino library manager, including examples.
Detailed Info about the library is in sinric pro repo: https://github.com/sinricpro/esp8266-esp32-sdk
Including examples and user documentation.
(Examples are not available for every device type now, but we're working on it)

* Much more device types available
* No dealing with json code

slimlesaurien, 2020-10-20:
hello sir I'm just starting out and I don't know how to do what you have to modify to be able to transfer the code thank you in advance for your help