kakopappa/sinric #146: Switch shows on GH but it doesn't turn on

VongolaVaro, 2018-11-07:
Hi kakopappa, everyone,

Thanks for reading, following your tutorial on how to add an ESP8266 Relay Switch (which is awesome and well written by the way), I made my account on Sinric (I'm a total noob on this, I'm learning), follow all the steps, I managed to get GH app to show the switch, the problem is that the ESP doesn't turn on or off, I used the wemos D1 example [below], the things I changed were: 

*Changed Api Key, SSID, PASSWORD*
#define MyApiKey "xxxxxxx" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "xxxxxxxxxxx" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "xxxxxxxxxxxxx" // TODO: Change to your Wifi network password

*Changed relayPin*
const int relayPin = D1;

*Changed device ID with the one I created*
void turnOn(String deviceId) {
  if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    
     digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH
  } 
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);    
  }     
}

void turnOff(String deviceId) {
   if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of first device
   {  
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     
     digitalWrite(relayPin, LOW);  // turn off relay with voltage LOW
   }
   else {
     Serial.print("Turn off for unknown device id: ");
     Serial.println(deviceId);    
  }
}

Those were the only things I changed from the code, when I plugged the NodeMCU the relay LED turned on, so I think I'm in the right path, but I can't change the relay status with the GH app.

Where do you think I'm having an error?
Thanks in advance

kakopappa, 2018-11-09:
did you set the mode in setup function for relayPin ? eg:  digitalWrite(relayPin, HIGH);

This is a simple example for relay controller.

```
int relayPin = 7;
void setup() {
  pinMode(relayPin, OUTPUT);
  digitalWrite(relayPin, HIGH);
}
void loop() {
  digitalWrite(relayPin, LOW);
  delay(3000);
  digitalWrite(relayPin, HIGH);
  delay(3000);
}
```

dorsi90, 2018-11-10:
I have the same issue.
My board works very well with alexa while google home set the lights on/off in sinric (i can see that state changes also in alexa app but the callback function is never called on the esp8266.

VongolaVaro, 2018-11-11:
> did you set the mode in setup function for relayPin ? eg: digitalWrite(relayPin, HIGH);
>
> This is a simple example for relay controller.
>
> ```
> int relayPin = 7;
> void setup() {
>   pinMode(relayPin, [...]

yes, but no luck so far! :'(

Jahnkeanater, 2018-11-23:
I'm having the same issue. The Google Home finds all my sinric lights but cannot control them. Alexa can control them flawlessly.

kakopappa, 2018-11-23:
Do you see the command from GH in the serial monitor when you ask GH to turn on/off a deivce?