kakopappa/sinric #147: sinric gets disconnected every 24 hours from google home.

RWeducation, 2018-11-09:
I'm having this issue that my sinric devices gets disconnected from google home and the only way to get them working again is to un-link and re-link the service. Is there a fix for this?

kakopappa, 2018-11-12:
@Fleker any idea?

Fleker, 2018-11-12:
Hey folks. I'm not aware of any outages on the smart home integration as a whole. There may be some issues with the project's integration with [request sync](https://developers.google.com/actions/smarthome/request-sync) if that's been implemented.

By disconnected, do you mean the devices are no longer controllable, or they disappear entirely from your list of devices?

kakopappa, 2018-11-13:
@Fleker elaborate the problem little bit.

Once you link the action everything works fines. Users can turn on/off devices using GoogleHome without a problem. After 24 hours or so when you ask GoogleHome to turn on/off a device, Google Home says Sinric is unavailable. When you un-link and link again it seems to work for next 24 hours. 

When Google Home says "Sinric is unavailable" i do not see any incoming requests from GoogleHome servers as well. (No errors in the server as well.) I assume at some point Google Home servers think Sinric is unavailable or is there a time limit ?

Normall, unless you add or delete a device, I do not send the sync request.

RWeducation, 2018-11-13:
Looks like enabling request sync solved the issue for me. Sonic is working for straight 3 -4 days  now.

kakopappa, 2018-11-13:
@RWeducation can you add more details about what you did ?

RWeducation, 2018-11-13:
@kakopappa i just followed the tutorial on how to setup Google Home and enabled homegraph API as stated in step 24.

Deimos1994, 2018-11-15:
It happens rarely to me, it happens maybe once a month or so (not a big deal for me). Why for some people it is 24 hours but for me it is about 1 month?