kakopappa/sinric #148: Thermostat and Google Home cannot report temperature below 0 degree Celsius.

Deimos1994, 2018-11-12:
Hi, 
I ask GH: **what temperature outside**
GH responds: **Cooling is set to 23, with a current temperature of 25.**
In Amazon Alexa app I see correct temperature (-2), but in GH app it is incorrect.
When temperature higher than 0 degree Celsius then it works fine.
Is it google's bug?
[image]

kakopappa, 2018-11-13:
could be a bug in the code. like if temp > 0 send it to GH.. it's never temp > 0 where i live. so i never thought about it 💯

Deimos1994, 2018-12-05:
@kakopappa, Wow, it works now, did you fix it? GH says: **It's currently -0.5 degrees.**

kakopappa, 2018-12-06:
Hello. Yes I believe I saw the issue and fixed it in the last update. Sorry
I forgot to update the issue

kakopappa, 2018-12-06:
I had a temperature > 0 then send the temperature to google condition.

It didn’t occur to me it can be less than 0 because it has never been less
than 15 where I live

Deimos1994, 2018-12-06:
@kakopappa, Thank you very much, I am closing the issue now.

closed 2018-12-06 by Deimos1994