kakopappa/sinric #149: Send brightness level to a PIN

CollaVinilica, 2018-11-12:
Hi, would it be possible to have an example code like the Wemos D1 Mini with Relay that uses a pin to set the brightness level of a white (single color) LED Strip using a MOSFET instead of using a relay? I've tried the examples but I'm unable to send the value to a pin instead of just printing it to the serial monitor. Could anyone help me?

CollaVinilica, 2018-11-24:
Someone? Please?

kakopappa, 2018-11-24:
@CollaVinilica 

dude your question is not clear.

I assume you are trying to control the brightness of the LEDs to turn it on /off

1. Do you want help with linking this to Alexa or with wiring it? 
Can you tell us what you have done so far? 

2. Is there a tutorial you follow to achieve this? share more details about the LED stripe you are using

I do not have a LED stripe. So I am just guessing here.  I found an example here. but it is RGB LED stripe. I assume it is working based on the same principals. 

https://learn.adafruit.com/rgb-led-strips/arduino-code
https://www.makeuseof.com/tag/connect-led-light-strips-arduino/

Use this example:
https://github.com/kakopappa/sinric/blob/master/arduino_examples/light_example2.ino#L122

based on the "value" you have to set the light strip brightness

CollaVinilica, 2018-11-24:
Hi, I'm sorry if I wasn't clear. I'm just trying to control the brightness level on a single color LED strip o any other LED light using a 12V input. Like you would do with a Philips Hue bulb, but instead I'm trying to use it your code. Light_example2.ino can print the brightness level on a serial monitor, but other than that, I would like to send that serial value to a PWM Pin on an ESP module. As a circuit I would do something like this https://i.stack.imgur.com/tfSQ1.png

kakopappa, 2018-11-25:
First, try this code.
http://yaab-arduino.blogspot.com/2015/01/driving-led-strip-with-arduino.html
https://programmingelectronics.com/tutorial-10-fade-an-led-with-pulse-width-modulation-using-analogwrite/

You might need a CAP to avoid flickering (https://electronics.stackexchange.com/questions/196344/led-strip-dimming)

Does it fade in/out? Can you get it to work? then things are easy from there.

then in the sketch create a function like this and pass the brightness

fade_in_out(int brightness) {
   // update the code from sample
}

.....
else if(action == "SetBrightness") {
	int brightness = json["value"];
	fade_in_out(brightness);	
}
..

CollaVinilica, 2018-11-25:
Hi, I already know to how to use PWM in the code, I just wanted some help to add the functionality on your code, like the relay example, but using a MOSFET to control the brightness level of a strip using PWM instead of a simple HIGH o LOW.

CollaVinilica, 2018-12-08:
So, can you help me?

servermcp, 2018-12-19:
else if(action == "action.devices.commands.BrightnessAbsolute") {
        
        int brightness = json ["value"]["brightness"];
        
        Serial.println("[WSc] brightness: " + brightness);
        analogWrite (rel1,brightness);

Its work for me

servermcp, 2018-12-19:
https://github.com/kakopappa/sinric/issues/169

Try it.

Addi100174, 2018-12-26:
Könnte jemand den code für set brightness komplett hier veröffentlichen , ich bekomme es nicht hin .

Addi100174, 2018-12-26:
Hier veröffentlichen bitte

servermcp, 2018-12-26:
For Google home with OTA and wifi mánager.

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <WiFiManager.h>
#include <WiFiClient.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

ESP8266WebServer httpServer(80);
WebSocketsClient webSocket;
WiFiClient client;
ESP8266HTTPUpdateServer httpUpdater;

#define MyApiKey "4ff18951-d947-4dff-ade1-7416a9058afd" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define PIN 2 // GPIO 2
#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

const char* host = "BMCupdate";

 
void turnOn(String deviceId) {
  if (deviceId == "5c05e7188d32c11d9e08ccce") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(PIN,HIGH);
  } 
 
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);    
  }     
}

void turnOff(String deviceId) {
   if (deviceId == "5c05e7188d32c11d9e08ccce") // Device ID of first device
   {  
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     digitalWrite(PIN,LOW);
   }
   
  else {
     Serial.print("Turn off for unknown device id: ");
     Serial.println(deviceId);    
  }
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch(type) {
    case WStype_DISCONNECTED:
      isConnected = false;    
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
      isConnected = true;
      Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
      Serial.printf("Waiting for commands from sinric.com ...\n");        
      }
      break;
    case WStype_TEXT: {
        Serial.printf("[WSc] get text: %s\n", payload);
        // Example payloads

        // For Switch  types
        // {"deviceId":"xxx","action":"action.devices.commands.OnOff","value":{"on":true}} // https://developers.google.com/actions/smarthome/traits/onoff
        // {"deviceId":"xxx","action":"action.devices.commands.OnOff","value":{"on":false}}

        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject((char*)payload); 
        String deviceId = json ["deviceId"];     
        String action = json ["action"];
        
        if(action == "action.devices.commands.OnOff") { // Switch 
            String value = json ["value"]["on"];
            Serial.println(value); 
            
            if(value == "true") {
                turnOn(deviceId);
            } else {
                turnOff(deviceId);
            }
        }
        else if(action == "action.devices.commands.BrightnessAbsolute") {
            
            int brightness = json ["value"]["brightness"];
            
            Serial.println("[WSc] brightness: " + brightness);
            analogWrite (PIN,brightness*10);
        }

        else if (action == "test") {
            Serial.println("[WSc] received test command from sinric.com");
        }
      }
      break;
    case WStype_BIN:
      Serial.printf("[WSc] get binary length: %u\n", length);
      break;
  }
}

void setup() {
  Serial.begin(115200);
  pinMode(PIN, OUTPUT);
  digitalWrite(PIN,HIGH);
  WiFiManager wifiManager;

  wifiManager.setTimeout(180);

  if(!wifiManager.autoConnect("BMC Device")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
}

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/"); //"iot.sinric.com", 80

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
   
   //OTA
   MDNS.begin(host);

  httpUpdater.setup(&httpServer);
  httpServer.begin();

  MDNS.addService("http", "tcp", 80);
  Serial.printf("HTTPUpdateServer ready! Open http://%s.local/update in your browser\n", host);
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  } 
    //OTA
  httpServer.handleClient();  
}

Addi100174, 2018-12-27:
Geht leider nicht

Addi100174, 2018-12-27:
Kann keiner helfen ?