kakopappa/sinric #156: change Email Address

GaryLittlemore, 2018-11-20:
I no longer have access to the email address that I used when I registered on the sinric website. I can still currently log in to my account but there isn't a function to be able to update my email address with my latest details.

kakopappa, 2018-11-20:
well sorry about that. Right now your best option is to keep it as it is.
Your email is being used to login only. I have to add the password reset
option and personal info changes features later. If you really want to
change it best option is to create a new account or drop me your email
address here I will change it when I get a chance(not sure when)

GaryLittlemore, 2018-11-21:
I’ll keep it as it is, thanks for the reply.

closed 2018-11-21 by GaryLittlemore