kakopappa/sinric #160: Rename devices via website

gizmocuz, 2018-11-27:
It would be great if there was an 'edit' button, where we could edit the device Name/Description in the web interface.
It's a feature that could help with issue https://github.com/kakopappa/sinric/issues/13

kakopappa, 2018-11-28:
Thanks. Started working on this already.

gizmocuz, 2018-11-28:
Great !! May i also suggest a small additional 'edit' option, 'alias'. Maybe it would be great if you could add 1 alias for the device name.

For example, i like to say "Set thermostat temperature to 20.5 degrees", someone else in my house likes to say "Set heating temperature to 20.5 degrees".
Currently i have to add two devices

"Turn on Living room" vs "Turn on Lights" is another example

kakopappa, 2018-12-01:
I have added editing device names support in release 1.4. Adding an "alias" will be supported in a later version. I will create a new issue for this.

closed 2018-12-01 by kakopappa