kakopappa/sinric #161: Sinric web server/socket/skill source

gizmocuz, 2018-11-27:
Would you be willing to share the source code of the skill ? (and webserver) ?

The reason for this is that i am interested in making a plugin for Domoticz if that is OK with you ?
(And i would like to have a backup of everything... )
(If you like you can reach me via infoATdomoticz(dot.)com)

Thanks for the great skill ! Working excellent !

kakopappa, 2018-11-28:
Sorry, I am not willing to share the source code or sell it.

If you can send me a sample code/prototype model, I am happy to integrate it to Sinric. 

Thanks.

gizmocuz, 2018-11-28:
@kakopappa , no problem !

As i was currently testing, I made a sketch/example for the ESP8266 that works perfectly now with your skill and Domoticz.
If it's OK i would like to make a pull request so you could merge this in the examples, it could be of benefit for others.

It could also serve as a example for non-domoticz users if you remove the domoticz related code ;)

kakopappa, 2018-11-28:
awsome. i can do that.

mayankraichura, 2018-11-29:
Hi All,

I've been working on a Sinric Node-Red module that supports Home Assitant and Domoticz. It's a rough cut and still a work in progress but its been working with Domoticz for over 15 days on RPi Model B. You are welcome to try and review/criticize it.

https://github.com/mayankraichura/node-red-contrib-sinric-ws

Thanks,
Mayank

kakopappa, 2018-11-29:
Good job. Thanks a lot for helping me to expand Sinric.

I will look at the repo and get back to you again. I still don’t know
what’s node-red and how it works :d

acostadaniel1, 2018-12-08:
hey Guys, really good job, to both of you, i'm using Sinric for a while with the esp8266 and now i found the node-red module, I know that's in development, but the debug node just provide the basic information to make working my small project, (google home + control some lights/devices at home)

Thanks