kakopappa/sinric #163: French skill

Nickduino, 2018-11-28:
Hi,

Please tell me I can help you port the skill to French :-)

kakopappa, 2018-11-29:
Hi, I have plans to supprot French after I complete Spanish. 
https://github.com/kakopappa/sinric/issues/94

Nickduino, 2018-11-29:
I should have searched a bit more, sorry.

Is that all that's needed to be translated? In the case of window shutters (it will be blinds for me), does the skill simply pass whatever you tell Alexa or do you need keywords.

Alternatively, would you share the source code of the skill, so that people can build it with their own amazon developer account?

kakopappa, 2018-11-29:
@Nickduino Yes that's all for now. 

For some reason it got rejected because somone at Amazon thought it was not correct and asked me to make few changes to the text and submit again. So I did. now we have to wait ... 

source code - Sorry I cannot share it with anyone because of security risks. I do not want anyone to exploit a bug and risk users. 

I am sure you can find  lot of example on internet already. Plus having the skill is not enough, You have to handle oauth requests (when linking)

kakopappa, 2018-11-29:
> I'll wait patiently, then. I wish Amazon would either allow to install
> foreign skills or implement a sort of closed beta testing to test skills
> before they are actually published.

Drop your email here I will send you a beta test invitation.

Nickduino, 2018-11-29:
stephane.bauder@gmail.com

Thanks!

Plarkass, 2018-12-25:
Hello,
I would also like to use French skills. 
Can I help you translate or do something?

kakopappa, 2018-12-26:
Hi, haven’t submitted it yet. I will do it tonight

Nickduino, 2018-12-26:
> [stephane.bauder@gmail.com](mailto:stephane.bauder@gmail.com)
> 
> Thanks!

Did you send me the beta invitation?

kakopappa, 2018-12-27:
submitted for testing. 

@Nickduino sent you an invite

Plarkass, 2018-12-27:
You can test to send it to me if needed.

Sent from Nine

Hum...never got it :-(

kakopappa, 2018-12-28:
won't let me change it after submission anymore

[image]

Nickduino, 2018-12-28:
Well, we'll have to wait, then

kakopappa, 2019-01-24:
Just got the confirmation. Skill is now live

Nickduino, 2019-01-24:
Not yet: https://www.amazon.fr/s/ref=nb_sb_noss?__mk_fr_FR=%C3%85M%C3%85%C5%BD%C3%95%C3%91&url=search-alias%3Dalexa-skills&field-keywords=Sinric

kakopappa, 2019-01-25:
now they do :)