kakopappa/sinric #165: Wemos D1 mini not receiving alexa commands

Mr-Moose2, 2018-11-29:
Hello:
I can log into the Sinric web site and test my device with and Alexa does discover the device, but when I tell Alexd to turn on the device, I don't receive any message.
Thanks Mrmoose2

kakopappa, 2018-11-29:
Are you using multiple Alexa profiles?

Mr-Moose2, 2018-11-29:
Im not sure what you mean my prohiles.

Mr-Moose2, 2018-11-29:
When I use the Alexa app I go to devices, switches. When I click on the shitch it says "Waiting for Sinric..."

Mr-Moose2, 2018-11-29:
It is working now. I just realized I had 2 iterations of the device. I deleated one and it is now working.
Thanks for your help and thanks for creating this site!

kakopappa, 2018-11-29:
https://www.amazon.com/gp/help/customer/display.html?nodeId=201628040

Sometimes when you have multiple house hold profiles, you have to use the
main account to install and link the account

Try unliking the skill and linking it again. I have the same setup at home.
Wemos D1 connected to a fan. I can turn on/off my fan(just checked)

Sorry I am AFK

closed 2018-11-30 by kakopappa