kakopappa/sinric #168: How do I use the speakers example?

The-Great-One, 2018-12-01:
I saw your code for speakers and i tried to use it, since i am not a programmer i tired to use some snippets of code from this https://github.com/chunter1/ESP8266_WLAN_speaker but i was unable to make it work, hope you could guide me on how to use this excellent piece of code. Thanks

kakopappa, 2018-12-03:
Sorry I have no idea. Speaker code is for something you can control using IR. I am not sure how ESP8266_WLAN_speaker  works. 

I guess you just have to handle the commands and control speaker

The-Great-One, 2018-12-07:
so how do I use your example code https://github.com/kakopappa/sinric/blob/master/arduino_examples/speakers_example.ino on an esp8266 like how do i define the pin outs and how do i wire the circuit for it