kakopappa/sinric #169: lights example for google home

servermcp, 2018-12-02:
Hi, thanks for your work, although I do not have much idea I managed to make the example of the switch in google home work !! Now I would like to be able to make the lights work, but even though I read and read, I can not even remotely understand how I could. Would you have an example like the switch?
thank you!!

kakopappa, 2018-12-03:
What kind of lights are you talking about? LED RGB light strip or bulb/lamp ?

Both can be turned on/off by a relay. Controlling the color is bit complicated. I have seen some bulbs can be controlled using IR and some using RF.

servermcp, 2018-12-03:
Hello, thanks for answering! Basically we want to control the brightness of a LED lamp through a pwm signal by an ESP analog pin on a MOSFET. Of the circuit to handle the high voltage, (here in Argentina we handle 220v) and the electronic part is taken care of by my friend.

[image]

servermcp, 2018-12-04:
else if(action == "action.devices.commands.BrightnessAbsolute") {
            
            int brightness = json ["value"]["brightness"];
            
            Serial.println("[WSc] brightness: " + brightness);
            analogWrite (rel1,brightness);

Hello!! I did that, I never imagined it but it worked! but obviously 100% (brightness = 100) is not 1023 and when I use the brightness "adjust brightness to 50%" it never shines again at 100%, in principle I can multiply brightness by 10 and it would be very close to the value and there would be almost no difference, but perhaps there is a more orthodox method to reach the correct values ​​(100% = 1023, 40% = 410, etc). Can you think of something? Thank you

closed 2019-01-31 by servermcp