kakopappa/sinric #170: Sinric for multiple apartments

innocuous, 2018-12-04:
Hi,
I want to implement sinric for 3 of my friends all staying in the same building, but in different apartments.
I would obviously NOT like to use the same sinric account to add all the devices (connected to multiple nodemcus).

I can create multiple sinric accounts and create devices to add to the individual nodemcus' in each apartment. Is there a way we can remotely add the device id to the code in nodemcu? Something like remote provisioning the device ids

Thanks

kakopappa, 2018-12-04:
if you have a single device connected to the MCU, you can just ignore the device id checking and run the command.

debadityadas, 2019-06-21:
How many node mcu can we connect to single sinric account by using the same API key but different device ID??

kakopappa, 2019-06-21:
no limitation