kakopappa/sinric #171: Disconnection issues

mluna0101, 2018-12-04:
Using this example: https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

Im getting this messages repeatedly  within hours:

[WSc] Webservice disconnected from sinric.com!
[WSc] Webservice disconnected from sinric.com!
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...
[WSc] Webservice disconnected from sinric.com!
[WSc] Webservice disconnected from sinric.com!
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...
[WSc] Webservice disconnected from sinric.com!
[WSc] Webservice disconnected from sinric.com!
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...
[WSc] Webservice disconnected from sinric.com!
[WSc] Webservice disconnected from sinric.com!
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...

Wifi connection is stable. Is this normal?

Deimos1994, 2018-12-04:
I have the same issue

```
When: December 4, 2018 at 04:57PM
Log: [WSc] Connected to sinric.com ...,
When: December 4, 2018 at 05:07PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 05:07PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 05:07PM
Log: [WSc] Connected to sinric.com ...,
When: December 4, 2018 at 05:09PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 05:09PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 05:09PM
Log: [WSc] Connected to sinric.com ...,
When: December 4, 2018 at 05:40PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 05:40PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 05:40PM
Log: [WSc] Connected to sinric.com ...,
When: December 4, 2018 at 06:13PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 06:13PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 06:13PM
Log: [WSc] Connected to sinric.com ...,
When: December 4, 2018 at 06:22PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 06:22PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 06:22PM
Log: [WSc] Connected to sinric.com ...,
When: December 4, 2018 at 06:28PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 06:28PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 06:28PM
Log: [WSc] Connected to sinric.com ...,
When: December 4, 2018 at 06:40PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 06:40PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 06:40PM
Log: [WSc] Connected to sinric.com ...,
When: December 4, 2018 at 06:40PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 06:40PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 06:40PM
Log: [WSc] Connected to sinric.com ...,
When: December 4, 2018 at 06:41PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 06:41PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 06:41PM
Log: [WSc] Connected to sinric.com ...,
When: December 4, 2018 at 07:37PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 07:37PM
Log: [WSc] Disconnected from sinric.com!,
When: December 4, 2018 at 07:37PM
Log: [WSc] Connected to sinric.com ...,

```

Server reboots or resets all connections. I opened sinric.com just after disconnection and saw just 10-200 devises online, then after reload it get back to normal 500-600 devices online.

kakopappa, 2018-12-05:
thanks guys.. checking this now

Deimos1994, 2018-12-05:
@kakopappa, Ok, thanks. I think it would be great to add some reasonable limits. For example, if someone uses setPowerStateOnServer in the loop, block them let's say for an hour, and notify them via e-mail. It is not fair that 1% of users use 80% of CPU time. Sinric works fine for me, (even with those disconnections) but what would happen when we'll have 1000-5000 users?... well we'll see, thanks for your work.

mluna0101, 2018-12-05:
Thank you @kakopappa 

> This seems to crash and restart the server. likely above issue. I fixed this
> 
> ClouldFlair seems to have some kind of a limitation for WebSocket connections when you are freeloading 

May I ask how you fixed it?. I don't now much about how does this web-services work, but I'd like to learn. Is sinric running in cloudflare? Is sinric independent from the Amazon cloud? How may I introduce myself to all these things you do to understand how it works? Can I run sinric in a local server?

Deimos1994, 2018-12-06:
Well, the problem still persists
```
When: December 6, 2018 at 03:08PM
Log: [WSc] Connected to sinric.com ...,
When: December 6, 2018 at 03:09PM
Log: [WSc] Disconnected from sinric.com!,
When: December 6, 2018 at 03:09PM
Log: [WSc] Disconnected from sinric.com!,
When: December 6, 2018 at 03:09PM
Log: [WSc] Connected to sinric.com ...,
When: December 6, 2018 at 03:09PM
Log: [WSc] Disconnected from sinric.com!,
When: December 6, 2018 at 03:09PM
Log: [WSc] Disconnected from sinric.com!,
When: December 6, 2018 at 03:09PM
Log: [WSc] Connected to sinric.com ...,
When: December 6, 2018 at 03:20PM
Log: [WSc] Disconnected from sinric.com!,
When: December 6, 2018 at 03:20PM
Log: [WSc] Disconnected from sinric.com!,
When: December 6, 2018 at 03:21PM
Log: [WSc] Disconnected from sinric.com!,
When: December 6, 2018 at 03:21PM
Log: [WSc] Connected to sinric.com ...,
When: December 6, 2018 at 03:23PM
Log: [WSc] Disconnected from sinric.com!,
When: December 6, 2018 at 03:23PM
Log: [WSc] Disconnected from sinric.com!,
When: December 6, 2018 at 03:23PM
Log: [WSc] Connected to sinric.com ...,
When: December 6, 2018 at 03:25PM
Log: [WSc] Disconnected from sinric.com!,
When: December 6, 2018 at 03:25PM
Log: [WSc] Disconnected from sinric.com!,
When: December 6, 2018 at 03:25PM
Log: [WSc] Connected to sinric.com ...,
```
It is 03:35PM now (my local time).
Sometimes it is stable for hours, and then something happens and it starts to disconnect.
(not a big deal for me, just letting you know).

Deimos1994, 2018-12-07:
@kakopappa It may sound crazy, but I can crash your server just by saying "hey google, set lights to blue" (or any other color). I tried it maybe 15 times, and it always crashes the server. ALWAYS! can you confirm?
It successfully sets lights to blue, and then the server crashes, WebSocket disconnects, and on sinric.com we get less than 100 devices online.

mluna0101, 2018-12-08:
@kakopappa thanks for taking the time to answer. I get all you said. But when I asked how may I introduce myself I actually meant where do you recommend me to start learning, like, by reading about Lambda?, by learning a specific programming language?, how did you start making sinric? do you have any background I can read about?

> You can have a local server but it make things bit complicated. (because you don't have a public ip).
That can be addressed through a DDNS.

@Deimos1994 as it works I actually stopped monitoring mine. First time I saw the disconnections I thought It was a problem in my side. Thanks for reporting though. Hope this thread can help make sinric better as it is a really nice service.

kakopappa, 2018-12-08:
@Deimos1994 Thanks I am trying to debug the issue on the server. Someone is flooding the server again with 

"action":"action.devices.commands.OnOff"

I have sent that person an email. I have to wait until he responds to take a look. 

@mluna0101 I was not sure whether you are a beginner or advanced person. 

Let me break this down.

Alexa Eco > 
You have to create a skill and configure it to use your server. This is simple

Alexa Lambada (Amazon cloud) > 
Lambada expression is written using NodeJS

Clouldflair > 
Just configuring it to use

Sinric > 
written using NodeJS

Your ESP module
Arduino

I like to recommend you to try a course from Udemy get exposure to Alexa Skill development.
https://www.udemy.com/comprehensive-alexa-skill-development-course/

Deimos1994, 2018-12-08:
@mluna0101 
> That can be addressed through a DDNS

Actually, it cannot.  If no public IP, means you only have a local IP address (provided by your ISP), for example, you can just have 192.168.X.X, and such IP cannot be accessible from the internet. DDNS can only help if you have a public IP that often changes.
But even if you have a public IP, to make it work (for google home), you will need to Set up an OAuth 2.0 server https://developers.google.com/actions/smarthome/create#setup-server
There are probably some workarounds exist, but it is all not that simple, especially for people like me who do not know much about server. And that is why Sinric exists.
@kakopappa, thanks! I cannot crash it now :)

mluna0101, 2018-12-08:
@Deimos1994 

> such IP cannot be accessible from the internet

In fact it can, that's why NAT and Port Forwarding exist. You can expose your private IP to the internet and access to your server through your public IP assigned by your ISP, you can even run different physical servers with different private IP addresses and access all of them through one public IP. The problem is when your public IP changes, you would have to find out what your new public IP address is and that's where DDNS comes handy.

You could also set up a VPN server and access your entire private network as if you were there.

@kakopappa Interesting. Thank you!, I'm gonna start by taking that course you recommended me.

Deimos1994, 2018-12-08:
@mluna0101 there is an ISP in my area that gives every customer IP like 192.168.2XX.XXX how am I going to make port forwarding? I don't have access to my ISP's router, so it is impossible to make Port Forwarding from my side. Yes, I have a public IP address, but that address is used by other customers at the same time. If I make a request to my public IP from the internet, that request simply does not reach my router... yes my ISP could forward that request to my local IP, but that is not what happens here. Your ISP probably gives you unique public IP, and you probably think that it is the case with all ISPs, but it is not.

shkdxb, 2019-06-23:
Hi Just Installed Sinric and having issue like disconnecting exactly on a minute. attaching my CPP file which is copy of sinric/arduino_examples/switch_example2.ino with little modifications
My serial monitor output as below:
WiFi connected. IP address: 192.168.1.3
0:00:00:06 [WSc] Service connected to sinric.com at url: /
0:00:00:06 Waiting for commands from sinric.com ...
0:00:01:00 [WSc] Webservice disconnected from sinric.com!
0:00:01:04 [WSc] Service connected to sinric.com at url: /
0:00:01:04 Waiting for commands from sinric.com ...
0:00:02:00 [WSc] Webservice disconnected from sinric.com!
0:00:02:00 [WSc] Webservice disconnected from sinric.com!
0:00:02:05 [WSc] Service connected to sinric.com at url: /
0:00:02:05 Waiting for commands from sinric.com ...
0:00:03:00 [WSc] Webservice disconnected from sinric.com!
0:00:03:00 [WSc] Webservice disconnected from sinric.com!
0:00:03:05 [WSc] Service connected to sinric.com at url: /
0:00:03:05 Waiting for commands from sinric.com ...
0:00:04:00 [WSc] Webservice disconnected from sinric.com!
0:00:04:00 [WSc] Webservice disconnected from sinric.com!
0:00:04:05 [WSc] Service connected to sinric.com at url: /
0:00:04:05 Waiting for commands from sinric.com ...
0:00:05:00 [WSc] Webservice disconnected from sinric.com!
0:00:05:00 [WSc] Webservice disconnected from sinric.com!
0:00:05:05 [WSc] Service connected to sinric.com at url: /
0:00:05:05 Waiting for commands from sinric.com ...
0:00:06:00 [WSc] Webservice disconnected from sinric.com!
0:00:06:00 [WSc] Webservice disconnected from sinric.com!
0:00:06:05 [WSc] Service connected to sinric.com at url: /
0:00:06:05 Waiting for commands from sinric.com ...
0:00:07:00 [WSc] Webservice disconnected from sinric.com!
0:00:07:00 [WSc] Webservice disconnected from sinric.com!
0:00:07:05 [WSc] Service connected to sinric.com at url: /
0:00:07:05 Waiting for commands from sinric.com ...
0:00:08:00 [WSc] Webservice disconnected from sinric.com!
0:00:08:00 [WSc] Webservice disconnected from sinric.com!
0:00:08:05 [WSc] Service connected to sinric.com at url: /
0:00:08:05 Waiting for commands from sinric.com ...
0:00:09:00 [WSc] Webservice disconnected from sinric.com!
0:00:09:00 [WSc] Webservice disconnected from sinric.com!
0:00:09:05 [WSc] Service connected to sinric.com at url: /
0:00:09:05 Waiting for commands from sinric.com ...

[main.zip](https://github.com/kakopappa/sinric/files/3318324/main.zip)

as you can see the server is disconnecting every minute and reconnecting to sinric.com

Pl help in identifying the issue, Thanks to @kakopappa for his excellent work on proving this service.

kakopappa, 2019-06-23:
@shkdxb 
I have been running your code for over 5 mins now (without stepper) seems to be fine. 

 ```

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h>      // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries (use the correct version)
#include <StreamString.h>
 
 
ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

#define MyApiKey "xx // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define WIFI_SSID "xx"                             // TODO: Change to your Wifi network SSID
#define WIFI_PASS "xx"                          // TODO: Change to your Wifi network password

#define DEVICE1 "xx" //TODO: Device ID of first device
#define DEVICE2 "xx" //TODO: Device ID of second device
#define DEVICE3 "xx" //TODO: Device ID of 3rd device

unsigned long UpTime = 0;

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId)
{ 

}

void turnOff(String deviceId)
{
   
}
 
void webSocketEvent(WStype_t type, uint8_t *payload, size_t length)
{
  switch (type)
  {
  case WStype_DISCONNECTED:
    isConnected = false;
    //Serial.print(TimeShowFormatted(millis()));
    Serial.printf(" [WSc] Webservice disconnected from sinric.com!\n");
    break;
  case WStype_CONNECTED:
  {
    isConnected = true;
    //Serial.print(TimeShowFormatted(millis()));
    Serial.printf(" [WSc] Service connected to sinric.com at url: %s\n", payload);
    //Serial.print(TimeShowFormatted(millis()));
    Serial.printf(" Waiting for commands from sinric.com ...\n");
  }
  break;
  case WStype_TEXT:
  {
    //Serial.print(TimeShowFormatted(millis()));
    Serial.printf(" [WSc] get text: %s\n", payload);
    // Example payloads

    // For Switch or Light device types
    // {"deviceId": xxxx, "action": "setPowerState", value: "ON"} // https://developer.amazon.com/docs/device-apis/alexa-powercontroller.html

    // For Light device type
    // Look at the light example in github

    DynamicJsonBuffer jsonBuffer;
    JsonObject &json = jsonBuffer.parseObject((char *)payload);
    String deviceId = json["deviceId"];
    String action = json["action"];

    if (action == "setPowerState")
    { // Switch or Light
      String value = json["value"];
      if (value == "ON")
      {
        turnOn(deviceId);
      }
      else
      {
        turnOff(deviceId);
      }
    }
    else if (action == "test")
    {
      Serial.println("[WSc] received test command from sinric.com");
    }
  }
  break;
  case WStype_BIN:
    Serial.printf("[WSc] get binary length: %u\n", length);
    break;
  default:
    break;
  }
}

void setup()
{
  Serial.begin(9600);
 

  WiFiMulti.addAP(WIFI_SSID, WIFI_PASS);
  Serial.println();
  Serial.print("Connecting to Wifi: ");
  Serial.println(WIFI_SSID);

  // Waiting for Wifi connect
  while (WiFiMulti.run() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  if (WiFiMulti.run() == WL_CONNECTED)
  {
    Serial.println("");
    Serial.print("WiFi connected. ");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);

  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000); // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop()
{
  webSocket.loop();

  if (isConnected)
  {
    uint64_t now = millis();

    // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
    if ((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL)
    {
      heartbeatTimestamp = now;
      webSocket.sendTXT("H");
    }
  }
}
```

 

[image]

shkdxb, 2019-06-24:
@kakopappa Thanks for quick response, it is strange - it is happening to me only. again i checked this  morning again and same disconnection happening exactly on a minute. 
Even i tried to reduce HEARTBEAT_INTERVAL to less than a minute but found no change in behaviour.
Can you pl direct me where i can look for solution for this. appreciate your time.
PS: i was also running my code without stepper motor connected

kakopappa, 2019-06-24:
Try starting an apn on your phone and connecting

shkdxb, 2019-06-24:
@kakopappa You mean Tethering using my mobile network for connection to your server instead of my home wifi?

kakopappa, 2019-06-24:
Yep

shkdxb, 2019-06-24:
yes, it works without disconnecting when connected through My mobile tethering network. I have gone through my routers setting and couldn't one which sets off the one minute time out. 
is there a setting in my router which disconnected network if there is no traffic?My router is ZTE F612W

closed 2020-03-24 by mluna0101