kakopappa/sinric #172: Spanish skill

jesusado, 2018-12-06:
Hi,

can I help you port the skill to Spanish?

It not be yet.

kakopappa, 2018-12-06:
It’s still being processed by Amazon

I sent them an email and this is the response I got

Hello,

Thank you for contacting us. I understand that "[Sinric]" was submitted a
little while ago and you'd like to know when you'll hear back about it.
I'll do my best to help you out with this today.

I checked our system and I can verify that the skill submission was
received and that our certification team is working through the testing
process. Please note that we aim to complete testing and provide feedback
within five business days of developers submitting their skill. If you have
not received a notification via the email associated with your developer
account by this time, please feel free to reach out again and we can follow
up for you.

jesusado, 2018-12-06:
five business days... mmmmm.. which year?

We will wait, thanks

kakopappa, 2018-12-07:
Dear Developer,

Thank you very much for having developed and submitted an Alexa Skill in
Spanish (MX) for certification. Like you, many other developers have
enthusiastically welcomed the possibility of developing Skills in Spanish
(MX) for Alexa. The Alexa certification team is working hard to process all
of the Skills that have been submitted. We appreciate your patience and,
while waiting for your skill to be tested, we encourage you to continue
developing new Skills for Alexa.

Thank you,

Alexa Skills Team
------------------------------

Estimado Desarrollador,

Muchas gracias por haber desarrollado y enviado una Skill Alexa para
certificación en español (MX). Otros desarrolladores como tú también han
dado la bienvenida a la posibilidad de crear Skills en español (MX). El
equipo de certificación de Alexa está trabajando duramente procesando todas
las Skills enviadas. Gracias por tu paciencia y, mientras esperas que se
hagan los tests de tu Skill, te animamos a que continues desarrollando
Skills para Alexa.

Gracias,

Alexa Skills Team

jesusado, 2018-12-07:
and the skill for Spanish (ES)?

kakopappa, 2018-12-20:
Skill is available now.
https://www.amazon.es/dp/B078RGYWQQ

closed 2018-12-20 by kakopappa