kakopappa/sinric #175: No output to relay, can I get some help from the experts, Please.

dhpete, 2018-12-17:
Sorry, I'm a NOOB. 
I have uploaded the code to my board ( tried WEMOS D1, ESP-01, and NodeMCU) entered all my info from SINRIC, 
I get confirmation on serial monitor WiFi connected-service connected to SINRIC-Waiting for commands. Use the test button on SINRIC, says all is well. Tell Alexa to turn on switch -ok, turning on switch. Shows in serial monitor (WSc) get text: (''deviceId''5myothernumbers4c'', action setpowerstate value ''on'' 
Same result if using the Alexa app
The relay never turns on! I have tried multiple pins, and still nothing. 
Could it be my code? Or my relay? I am using these   
https://www.amazon.com/gp/product/B00XT0OSUQ/ref=od_aui_detailpages00?ie=UTF8&psc=1
I could just be an idiot and am missing something. Any help would be great! Here is my code

/*
 Version 0.3 - March 06 2018
*/ 

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey ".........." // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "............" // TODO: Change to your Wifi network SSID
#define MyWifiPassword ".........." // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 
#define pinMode(pin, OUTPUT)
uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void setPowerStateOnServer(String deviceId, String value);
void setTargetTemperatureOnServer(String deviceId, String value, String scale);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "5.............1") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(D6, HIGH);
  } 
 else if (deviceId == " 5axxxxxxxxxxxxxxxxxxx") // Device ID of second device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    
  }
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);    
  }     
}

void turnOff(String deviceId) {
   if (deviceId == "5c139eb47892dc7d78e47da1") // Device ID of first device
   {  
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     digitalWrite(D6, LOW);
   }
   else if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of second device
   { 
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
  }
  else {
     Serial.print("Turn off for unknown device id: ");
     Serial.println(deviceId);    
  }
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch(type) {
    case WStype_DISCONNECTED:
      isConnected = false;    
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
      isConnected = true;
      Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
      Serial.printf("Waiting for commands from sinric.com ...\n");        
      }
      break;
    case WStype_TEXT: {
        Serial.printf("[WSc] get text: %s\n", payload);
        // Example payloads

        // For Switch or Light device types
        // {"deviceId": xxxx, "action": "setPowerState", value: "ON"} // https://developer.amazon.com/docs/device-apis/alexa-powercontroller.html

        // For Light device type
        // Look at the light example in github
          
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject((char*)payload); 
        String deviceId = json ["deviceId"];     
        String action = json ["action"];
        
        if(action == "setPowerState") { // Switch or Light
            String value = json ["value"];
            if(value == "ON") {
                turnOn(deviceId);
            } else {
                turnOff(deviceId);
            }
        }
        else if (action == "SetTargetTemperature") {
            String deviceId = json ["deviceId"];     
            String action = json ["action"];
            String value = json ["value"];
        }
        else if (action == "test") {
            Serial.println("[WSc] received test command from sinric.com");
        }
      }
      break;
    case WStype_BIN:
      Serial.printf("[WSc] get binary length: %u\n", length);
      break;
  }
}

void setup() {
  Serial.begin(115200);
  
  WiFiMulti.addAP(MySSID, MyWifiPassword);
  Serial.println();
  Serial.print("Connecting to Wifi: ");
  Serial.println(MySSID);  

  // Waiting for Wifi connect
  while(WiFiMulti.run() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  if(WiFiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.print("WiFi connected. ");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

// If you are going to use a push button to on/off the switch manually, use this function to update the status on the server
// so it will reflect on Alexa app.
// eg: setPowerStateOnServer("deviceid", "ON")

// Call ONLY If status changed. DO NOT CALL THIS IN loop() and overload the server. 

void setPowerStateOnServer(String deviceId, String value) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["deviceId"] = deviceId;
  root["action"] = "setPowerState";
  root["value"] = value;
  StreamString databuf;
  root.printTo(databuf);
  
  webSocket.sendTXT(databuf);
}

//eg: setPowerStateOnServer("deviceid", "CELSIUS", "25.0")

// Call ONLY If status changed. DO NOT CALL THIS IN loop() and overload the server. 

void setTargetTemperatureOnServer(String deviceId, String value, String scale) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["action"] = "SetTargetTemperature";
  root["deviceId"] = deviceId;
  
  JsonObject& valueObj = root.createNestedObject("value");
  JsonObject& targetSetpoint = valueObj.createNestedObject("targetSetpoint");
  targetSetpoint["value"] = value;
  targetSetpoint["scale"] = scale;
   
  StreamString databuf;
  root.printTo(databuf);
  
  webSocket.sendTXT(databuf);
}

Please help :)

kakopappa, 2018-12-17:
You are missing the

 pinMode(D6,OUTPUT);

In the setup()

dhpete, 2018-12-17:
Thanks. I placed the line here....
/*
 Version 0.3 - March 06 2018
*/ 

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "4f39ac6c-b804-4c74-9b99-ab37227905f8" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "iLive_ISWFV387_CC6C" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "allisonpayge399" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 
#define pinMode(D6,OUTPUT);
uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void setPowerStateOnServer(String deviceId, String value);
void setTargetTemperatureOnServer(String deviceId, String value, String scale);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "5c17b88e80767b3fcb697f4c") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(D6,LOW);

Is this correct? Still no output :(

dhpete, 2018-12-17:
Also this is what serial monitor looks like
[image] 

Sorry for the trouble. This is so simple I should be able to get it, but its driving me crazy :)

kakopappa, 2018-12-18:
#define pinMode(pin, OUTPUT) <----- that is wrong

`....
void setup() {
  Serial.begin(115200);

  pinMode(pin, OUTPUT); 
  ....
}
`

kakopappa, 2018-12-18:
example
https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example2.ino

dhpete, 2018-12-18:
Still no luck :( Moved pin to D2, relay turns on while programming then off when loaded. Tried voice commands and the switch button on app. Serial is receiving commands, but does not trigger relay. I must be the dumbest person in the world, because I just cant figure this out. Sorry for bothering you guys.

kakopappa, 2018-12-18:
Have you tried basic relay on/off sketch?

dhpete, 2018-12-18:
Like the blink sketch? Yes, I have and the relay cycles like it should.

kakopappa, 2018-12-18:
according to your serial monitor output, it says "turn on for unknown device id": So you have not added the device id to if condition.

1. get the device id from Sinric website and update

2. try changing to this

{
Serial.print("Turn off for unknown device id: ");
Serial.println(deviceId);
digitalWrite(relayPin1, HIGH); // turn on relay with voltage HIGH
}

dhpete, 2018-12-18:
I'll give it a try. Thanks for your help.

dhpete, 2018-12-19:
Got it working!!! Thanks for all the help, you are the greatest!!!

closed 2018-12-20 by kakopappa