kakopappa/sinric #178: Smartlock not visible as lock but as swith!

teumaas, 2018-12-20:
Hi as the title explains, i added a device that should be seen as a smartlock but is exactly seen as a switch.
I have tried to add more devices to see if it was my problem but the other devices are shown correctly in the Google Home app. Any idea how to solve this?

teumaas, 2018-12-20:
Thank you so much! Keep up the good work, i love your project! 👍

teumaas, 2018-12-23:
Hi again! Any idea when your fixing this? Sorry for asking, thank you. :)

kakopappa, 2018-12-24:
likely after the holidays

kakopappa, 2018-12-28:
Sorry about the delay. 

By looking at the Google Home documentation, it does not support Lock devices yet. All other types of devices are there.
https://developers.google.com/actions/smarthome/guides/

that's why I have picked the Outlet icon which is the closest thing to Lock/Unlock

teumaas, 2019-01-07:
Hmm, sad i have to wait than. Thanks for the response! I getting annoyed at Google for not already implementing this.

FufsowyFufs, 2019-08-27:
Google added a lock, garage door, gate, window, shutter, pergola, regular door device types. Surely you could use one of those types to set as a smart lock.