kakopappa/sinric #179: Amazon Alexa app is not reporting thermostat temp

emnik, 2018-12-21:
Hi kakopappa!
The Amazon Alexa app is showing the thermostat sinric device which is working as expected, it shows the on off button which is also working but when I click to get to the thermostat page is showing waiting sinric... And stays there showing the 3 dots for waiting. At least a month back it was showing the temperature... An idea what is happening? I already tried removing the thermostat and adding a new one from sinric but the result is the same...

emnik, 2018-12-24:
Any idea why is this happening? Does it happen to all  or is it only me?

The-Great-One, 2018-12-25:
> Hi kakopappa!
> The Amazon Alexa app is showing the thermostat sinric device which is working as expected regarding voice commands and responses, it shows the on off button which is also working but when I click to get to the thermostat page is showing waiting sinric... And stays there showing the 3 dots for waiting. At least a month back it was showing the temperature... An idea what is happening? I already tried removing the thermostat and adding a new one from sinric but the result is the same...

Hello emnik, 
                     Could you share the code you used to make the thermostat. I have been trying to use this code but since i have no programming skill i am unable to do so. Thanks!

kakopappa, 2018-12-28:
@emnik 

Finally got around to checking it. Seems to be fine. Is it still same for you

[image]

emnik, 2018-12-30:
Nope... Still the same here. If i ask for the thermostat's temperature
Alexa reports the temperature correct but the app is still showing the 3
dots and says waiting for sinric ..

kakopappa, 2018-12-31:
@emnik 
Can you check now, please?. 

Made a few changes and tested in my dev environment again. Is it less than 0 degress where you live now?

emnik, 2018-12-31:
> Oh. May be it’s needs mode. Can you set the mode and check again

I'll be away from home till next week... When I'll get back I'll try it.
Happy new year by the way ;-)

kakopappa, 2018-12-31:
Same to you!

kakopappa, 2019-01-05:
This issue has been a very frustrating one to fix. It used to work then did not. Perhaps it's the same reason why it stopped working for you as well. I think I spent over 6 hours trying to figure out what's wrong.  Response to Alexa is correct but it just refuses to show the updated temperature settings on the app. After few hours of changing the responses to Alexa managed to fix it. It seems Amazon is very picky when it comes to reading temperature when the thermostat is in different modes.

Check again. If it does not work you may have to delete the device and create a new one  @(perhaps changing the name if possible)  Alexa seems to remember the previous settings and seems to mess up sometimes (may be in cache still)

emnik, 2019-01-05:
Yes now it is working! I indeed needed to delete the device and add a new one with a different name. (I've done that also in the past without success, so the latest changes you did where good! ;-)

closed 2019-01-05 by emnik