kakopappa/sinric #180: Led dimmer hilfe , bekomme es nicht hin . Beispielcode bitte

Addi100174, 2018-12-27:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Addi100174, 2018-12-27:
Led dimmee ich brauche hilfe hat jemand ein code der geht

JanB97, 2019-01-09:
Die Wahrscheinlichkeit, dass dir jemand hilft, steigt drastisch, wenn du es in englisch postest und ein bisschen mehr Kontext dazu gibst.
Zum Beispiel:
Wie ist die Led (oder LEDs, Streifen?) angebunden.
Was hast du bereits?

So wie das hier steht, wird dir niemand helfen können, weil niemand weiß, was du genau willst.
Mal abgesehen davon, dass Licht dunkler werden soll.