kakopappa/sinric #181: smartlock in alexa not responsive

l0eke, 2018-12-28:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

l0eke, 2018-12-28:
Smartlock is discovered in alexa app but when connecting to it, alexa si trying to communicate with sinric. Always showing the message 'waiting for sinric'. All other devices like lights and switches are working correctly. Any suggestions? Thank you and great work man!

kakopappa, 2018-12-28:
whaaaat? i just restarted the server to fix another issue. 

can you check now ?

kakopappa, 2018-12-28:
tanks. i found the issue and fixed it. Should be ok now

l0eke, 2018-12-28:
Old ones seem to work!
New ones (create 10 minutes ago) not responsive but I’ll wait a bit for those 😊

Nice project man!!

kakopappa, 2018-12-28:
Thanks.  I was monitoring requests for few mins now.

The problem is once you create a lock device, the server has no idea whether it's locked or unlocked. So when the Alexa requests for the state, it does not send the lock state. I changed the code to send Unlocked states if you have not set the state before. After first lock / unlock command, it should be in sync

kakopappa, 2018-12-28:
> Seems to work now, commands are transferring from alexa to sinric.
> I do receive a message from Alexa ‘backdoor is not responding’ although
> command is send to sinric.
> Any ideas why Alexa is saying that it is not responding although message
> is going through?
>
> Thx!
> DVDK

likely response back to Alexa is wrong. I will check this tomorrow and let
you know

l0eke, 2018-12-28:
NP man, thx for your time!!